bigtreetech/BIGTREETECH-TouchScreenFirmware #1: Wrogn icons display and bitmap update stop @ 50/66 SD icons files

kattivik76, 2019-07-17:
I have make a new .bin file from source, it's include TFT_SD or OnboadSD printing functions, details:

Hardware version: BIGTREETECH-TFT35-V1.2
Firmware version: BIQU_TFT35_V1.2.20.bin (previously -> BIQU_TFT35_V1.2.13.x.bin but same issues)

I have used the complete and original bmp files folder from github [https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/new](url)

the issues is:

1. no correct icons display in "print" menu, after start print from TFT_SD or OnboardSD, the "printing" menu don't display some icons, display the white square

[image]

2. in this "printing" menu is't present the "back" icon button to return on main page, after printing a file from SD, the TFT display can be reset "poweroff - poweron" to return on the main page.

[image]

3. no correct icons display in "Babystep" 

[image]

and

[image]

`MENUITEMS babyStepItems = {
//   title
  LABEL_BABYSTEP,
//icon                        label
 {{ICON_DEC,                  LABEL_DEC},
  {ICON_BACKGROUND,           LABEL_BACKGROUND},
  {ICON_BACKGROUND,           LABEL_BACKGROUND},
  {ICON_INC,                  LABEL_INC},
  {ICON_NOZZLE,               LABEL_NOZZLE},
  {ICON_01_MM,                LABEL_01_MM},
  {ICON_NORMAL_SPEED,         LABEL_VALUE_ZERO},
  {ICON_BACK,                 LABEL_BACK},}
};`

4. when firmware updating from TFT_SD, the updating bmp files process whas display some icons pics and count of total icons, stop any bmp icons update whet OS reach 50 icons of 66 icon present. the complete firmware update continue without any error.

5. I have try to update only the last bmp files like:

-  OnTFTSD.bmp
- OnBoardSD.bmp
- U_Disk.bmp

### the updating process terminate without updating any new icons and without any errors

bigtreetech, 2019-07-18:
hi，if you are using ‘BIQU_TFT35_V1.2.20.bin’， please copy this 3 icons to SD Crad path “TFT35/bmp”
[image]

robbycandra, 2019-07-22:
It is because ICON NAME enum and BMP name is not sync.

kattivik76, 2019-07-22:
> hi，if you are using ‘BIQU_TFT35_V1.2.20.bin’， please copy this 3 icons to SD Crad path “TFT35/bmp”
> [image]

### Hi, i have try to move all file in TFT35/dmp and TFT35/font, now work fine!!!!

please update the howto documentation because the destination folder and bmp color deep isn't correct:

1- need to put all bmp files in SD_root/TFTxx/bmp folder
2- need to put all font files in SD_root/TFTxx/fontfolder
3- need to put new firmware update in SD_root folder
4- correct folder structure for TFT35 is:

```
SD_root/BIQU_TFT35_V1.2.20.bin "new firmware update"
SD_root/TFT35/bmp "all icons for 3.5" display"
SD_root/TFT35/font "all fonts for 3.5 display"
```

note, the SD_root/TFTxx folder is:
1- TFT35 >> 3,5" display
2- TFT28 >> 2.8" display
3- TFT24 >> 2.4" display

robbycandra, 2019-07-22:
Its already written in the last paragraph of readme.md

[image]

guruathwal, 2019-11-22:
@kattivik76 please close the issue if your problem is solved.

guruathwal, 2020-01-19:
@bigtreetech please close this issue

closed 2020-03-14 by bigtreetech