bigtreetech/BIGTREETECH-TouchScreenFirmware #10: Cleanner history with rebase

robbycandra, 2019-07-24:
Dear @bigtreetech , it is cleaner if PR not merged but rebased.

thisiskeithb, 2019-08-03:
Yes, please!

robbycandra, 2019-08-03:
Hello @thisiskeithb , nice to meet you here.

thisiskeithb, 2019-08-03:
You too! My TFT24 finally cleared customs, so now I'm busy putting in PRs :)

robbycandra, 2019-08-03:
I am still slow. Wait and see where the wind blow.

robbycandra, 2019-08-05:
@thisiskeithb , for me, its like split version of MKS Robin TFT. 
We need a big map on how this UI will talk with Marlin.
Do you have any idea about it ?

thisiskeithb, 2019-08-05:
The TFT side is just like the MKS TFTs since it's basically a standalone controller that sends gcodes back to Marlin.

For the 12864 emulation, it's just like the standard `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`/uses the standard Marlin UI.

robbycandra, 2019-08-05:
I make this Screen as Extensible UI. I love it because it has Encoder Button. Maybe i will rebuild the firmware.

thisiskeithb, 2019-08-05:
I'd be curious to see it! Have a link? :)

robbycandra, 2019-08-05:
It still in very early development. 
First thing I do is make a status screen like UltraLCD. 

BTW. we can also use code from here to build the "missing" MKS Robin TFT code

robbycandra, 2019-08-16:
[image]

thisiskeithb, 2019-08-16:
> this is what happened to my screen.

It's fine if you ignore the GUI side. 🙂But really, this probably won't change.

vMeph, 2019-08-18:
sorry to jump in @robbycandra and @thisiskeithb but it seems this is the only active issue lol
was wondering if this TFT 35 at some point will have some basic display like the reprap lcd
like first screen show x y z position temp of bed and hotend fan  and then yes go into menus
thats what seems to make sense in this LCD to have
at this point it seems that the  TFT firmware is missing alot stuff, i wanted to move my machine to have this LCD but  im waiting to see if the firmware becomes a litle more useful

robbycandra, 2019-08-18:
@vMeph , Marlin firmware already have something called Extensible UI that will take care communication between User Interface and Marlin. But it still need more work And The work must be done in LCD firmware and Marlin Firmware.

guruathwal, 2020-02-20:
@robbycandra Should this be closed now ?

robbycandra, 2020-02-21:
Yes Close

guruathwal, 2020-02-21:
Then please close it.

closed 2020-03-14 by bigtreetech