bigtreetech/BIGTREETECH-TouchScreenFirmware #12: Firmware update "stuck"

TVAVAE, 2019-07-30:
When putting BIGTREE_TFT24_V1.1.21.bin and the TFT24 folder alone at the root of the SD and re-powering. The screen says: 

Program size: 82kb
Updating:  %

And stays like this the whole time.

doughy2002, 2019-08-19:
I had this problem.  Change of memory card solved the problem

TVAVAE, 2019-08-19:
Indeed, changed +- 8 different cards and finally worked...

poisson21, 2019-09-10:
I found that on a exp1 exp2 config, no serial(for my case,didn't test with). to be able to flawlessly flash new firmware you have disconnect exp2 to make it work. I think the feedback of the "sd card presence" pin interfere with the upload.

nmavor, 2019-10-15:
@poisson21 workaround work for me (pull exp2 out)

mantoles, 2019-10-15:
This whole experience is pathetic.  Does not recognize the printer for me and only works in touch screen mode not serial too many bugs

closed 2020-03-14 by bigtreetech

hillsh, 2020-06-10:
Disconnecting EXP2 worked for me with a 32GB card.