bigtreetech/BIGTREETECH-TouchScreenFirmware #15: Russian Language

dmiart, 2019-08-01:
The Russian language no longer works, but it did with the original firmware that came with the LCD display. I noticed it was commented out in the code and the language_ru.h has weird encoding(in the source code) instead of the UTF-8. And additional related issue: the font for Russian was uneven. Some letters higher or lower than they should be. I am assuming its some sort problem with font itself? Is there any way to load custom .fon fonts? Where do we get them from. Thank you.

dmiart, 2019-08-01:
Found a post from old tft repo explaining some of those. 
This fork by Acenotass contains a better fonts for our screens. Even if you are gonna be using English as your language, its still better looking than the original one. 

https://github.com/Acenotass/BIGTREETECH-TFT35-V1.2/tree/master/firmware/Copy%20to%20SD%20Card%20root%20directory%20to%20update/TFT35/font  

and http://corosan.ru/sgfed/    Can be used to edit those bitmap fonts.

NukeAOSP, 2019-09-07:
We want russian lang, please =(

guruathwal, 2019-11-22:
@dmiart The Russian language is now added in the firmware.. please close the issue.

closed 2019-11-22 by dmiart