bigtreetech/BIGTREETECH-TouchScreenFirmware #19: What happened to WiFi support?

Strat00s, 2019-08-10:
Pretty much what the title says. One the FW my display came with (tft35 v2.0), there was a wifi option (which did absolutely nothing, but it was there). Now, this option totally disappeared.
So, will the WiFi support be added back in?

AustrianBossFan, 2019-08-31:
following.

i bougt a tft35 2.0. so whats the plan here? i want to use the wifi option, that was one of the reasons why i choosed this display...

jimmyeao, 2019-10-29:
> dont know, whats with the 2.0 - but if you are talking about the 3.0 - you need to get a ESP01S device, flash it with ESP3D, flash latest fw-version and plug the wifi-device into the tft - done.

Which version - ESP8266 core version or ESP32 core version. Is this the correct FW https://github.com/luc-github/ESP3D

p1ngb4ck, 2019-10-29:
Does not work. I did exactly that, but the data port cannot be used, since
it is just 2 unstable. Uploading of files to the tft-SD doesn't work
either. There have to be some problems in the tft firmware. I tinkered
around a lot with buffer sizes and baud rate - to no avail. As soon as you
start a print, it will stop on different progress points very soon after
the print starts

jimmyeao, 2019-10-29:
> Does not work. I did exactly that, but the data port cannot be used, since it is just 2 unstable. Uploading of files to the tft-SD doesn't work either. There have to be some problems in the tft [...]

I have that issue printing from the TFT SD card without WiFi. Are you able to print from the TFT SD  card with no issues otherwise?

closed 2020-08-14 by bigtreetech