bigtreetech/BIGTREETECH-TouchScreenFirmware #20: [Request] BLTouch support

Titi9237, 2019-08-14:
Hi,

Is there a plan to get a BLTouch/3DTouch support with the screen ?

Regards

DustyProp, 2019-09-11:
Bump ^ U was wondering the same thing. I would like to be able to use the home button and have it run the home with the BLTouch, not run off the edge of my bed and low enough to crack the glass if I let it go with out power cycling the printer. Good thing I have a pi that will run the G28; G29.

spilz87, 2019-09-11:
@Titi9237 The idea is to add buttons to reset/retract/down the bltouch ?

DustyProp, 2019-09-11:
Then why bother having the auto home button as part of the firmware? I don't see the point if adding buttons was the main design feature.

spilz87, 2019-09-11:
@DustyProp for me it’s different needs as you have on marlin lcd 
Homing + leveling as automatic is a “good” button.
Resolve an issue with bltouch without returning to lcd12864 mode can be a good idea, more over when we know that changing mode affect Marlin (because of using rotary encoder and because returning to touchscreen send G-code to marlin and init SD card)

Titi9237, 2019-09-12:
> @Titi9237 The idea is to add buttons to reset/retract/down the bltouch ?

Yes, it could be useful to add these buttons to the "homing screen".

spilz87, 2019-09-12:
Actually it’s not too difficult, the real need is icons

guruathwal, 2019-11-22:
@Titi9237 BLtouch feature has been added through PR #87 . You can close the issue now.

closed 2019-11-22 by Titi9237