bigtreetech/BIGTREETECH-TouchScreenFirmware #24: Unable to update the firmware

EnnajimiYassine, 2019-08-16:
I'm unable to update the firmware of my board tft35 ( hardware : V1.2, firmware : 1.2.6)

when i put the firmware on the SD card and reset nothing happen ( no firmware update), but when i put the bmp and font they get updated ( but still not the firmware)

Can please some one help ?

Doom6666, 2019-08-24:
same problem with TFT35 V2.0 .... only icons and fonts are updated :-(

EnnajimiYassine, 2019-08-25:
I renamed it to BIQU_TFT35.bin and it worked

Doom6666, 2019-08-25:
for me ..... renaming it to BIQU_TFT35.bin ... no change .... still firmware upgrade has not taken place :-(

EnnajimiYassine, 2019-08-25:
you tried  BIGTREETECH_TFT35.bin ?

Doom6666, 2019-08-25:
yes, I tried a lot of file names, including BIGTREETECH_TFT35 ... the upgrade was still unsuccessful. I'm clueless. Installation instructions absolutely useless :-(

Sthopeless, 2019-08-25:
I am also facing the same problem, cant upgrade TFT35 firmware. It updates the icons and fonts but not the firmware.
I have tried many names combinations.

AnHardt, 2019-08-25:
Connect a ST-link. Read out the bootloader and save. Open the file in a HEX-editor. Search for readable strings containing ".bin" and ".cur". And you'll get the right name- completely without guessing.

Msnejbjerg, 2019-08-27:
Same issue here with my new TFT35 2.0 running 2.0.20. I can update the TFT35 folder but I can't install new firmware

Doom6666, 2019-08-27:
> Same issue here with my new TFT35 2.0 running 2.0.20. I can update the TFT35 folder but I can't install new firmware

Unfortunately, I did not get a simple answer either here or directly from BIQU. It looks like product support isn't worth anything.

Doom6666, 2019-08-28:
> please download in here [https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update\](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update%5C)
> TFT32 V1.0 firmware name BIQU_TFT35_V1.0.22.bin
> TFT32 V1.1 firmware name BIQU_TFT35_V1.1.22.bin
> TFT32 V1.2 firmware name BIQU_TFT35_V1.2.22.bin
> TFT32 V2.0 firmware name BIQU_TFT35_APP1_V2.0.22.bin
> TFT28 firmware name BIQU_TFT28_V1.0.22.bin
> TFT24 firmware name BIGTREE_TFT24_V1.1.22.bin

I have already understood this, but after inserting the BIQU_TFT35_APP1_V2.0.22.bin file on the SD card, inserting the card into the TFT35 V2.0 and pressing the reset button, the firmware simply does not update ...

Msnejbjerg, 2019-08-28:
> > please download in here [https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update%5C)
> > TFT32 V1.0 firmware name BIQU_TFT35_V1.0.22.bin
> > TFT32 V1.1 firmware name BIQU_TFT35_V1.1.22.bin
> > TFT32 V1.2 firmware name BIQU_TFT35_V1.2.22.bin
> > TFT32 V2.0 firmware name BIQU_TFT35_APP1_V2.0.22.bin
> > TFT28 firmware name BIQU_TFT28_V1.0.22.bin
> > TFT24 firmware name BIGTREE_TFT24_V1.1.22.bin
> 
> I have already understood this, but after inserting the BIQU_TFT35_APP1_V2.0.22.bin file on the SD card, inserting the card into the TFT35 V2.0 and pressing the reset button, the firmware simply does not update ... Paradoxically, when you insert the TFT35 directory with icons and fonts, Updating icons and fonts runs without a problem.

That is exactly the same issue I have. It doesn't matter if I use the precompiled or if I compile it myself. I have tried several different filenames including the ones suggested in this thread. I bought two of these displays and it is the same problem with both

Msnejbjerg, 2019-08-28:
> Do you have a burner like stlink? Maybe I can send you a bootloader and try to reburning the bootloader?

I have a couple of nucleo boards with integrated stlink that I should be able to use. I'll give it a try

Doom6666, 2019-08-28:
> Do you have a burner like stlink? Maybe I can send you a bootloader and try to reburning the bootloader?

I have ST-link V2 ordered, but not yet arrived .... I can try it afterwards

AnHardt, 2019-08-28:
Is it expected the bootloaders rename the *.bin files to *.cur when flashed?

Msnejbjerg, 2019-08-28:
Succes! After uploading the new bootloader it worked!

EnnajimiYassine, 2019-08-28:
@AnHardt it doesn't change the firmware to .cur you will need to delete it or rename it afterward if you don't want to reflash your board at each reboot ( at least for tft35 v1.2)
@Msnejbjerg can you please tell us what firmware name did you use ? It may help the others

Msnejbjerg, 2019-08-28:
@EnnajimiYassine I used the boot loader first and flashed it with stink utility
[image]
[image]

[image]

And then I just used the precompiled file: BIQU_TFT35_APP1_V2.0.22.bin

EnnajimiYassine, 2019-08-28:
Thanks for sharing the info! It may help the others

bigtreetech, 2019-08-28:
please download in here https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update\
TFT32 V1.0  firmware name BIQU_TFT35_V1.0.22.bin
TFT32 V1.1  firmware name BIQU_TFT35_V1.1.22.bin
TFT32 V1.2  firmware name BIQU_TFT35_V1.2.22.bin
TFT32 V2.0  firmware name BIQU_TFT35_APP1_V2.0.22.bin
TFT28          firmware name BIQU_TFT28_V1.0.22.bin
TFT24          firmware name BIGTREE_TFT24_V1.1.22.bin

bigtreetech, 2019-08-28:
Do you have a burner like stlink? Maybe I can send you a bootloader and try to reburning the bootloader?

bigtreetech, 2019-08-28:
[BIQU_TFT35_V2.0_bootloader.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/3549545/BIQU_TFT35_V2.0_bootloader.zip)
here is the bootloader for TFT35_V2.0, burning it to flash address 0x08000000 is ok.
thanks a lot.

Juju-Dudu, 2019-09-02:
Ok, cool ! I have the same issue, but how can i do if i don't have any flashing board... I presume i'm not alone in this case... I'm a noob in programming, i never do that. Can I do it with an Arduino Uno ? Don't laugh, i'm a really noob !!!

petrik0621, 2019-09-03:
I need help too. :/

EnnajimiYassine, 2019-09-03:
@bigtreetech I think you should post some firmware names, so those who don't have an stlink can update their firmware. Because from what I understand we're supposed to update it with sd card (that's what your GitHub says) and not via stlink ....

GenaUser, 2019-09-07:
I have a same problem as @blindMoe with my TFT24-V1.1. 
After upload, I cannot get passed the calibration screen. It just says that calibration failed and starts the calibration sequence again...
I tried to use the master compiled for TFT24. Same problem.
Screen unusable.

poisson21, 2019-09-10:
On myy tft 24 v1.1 I found that on a exp1 exp2 config, no serial(for my case,didn't test with). to be able to flawlessly flash new firmware you have disconnect exp2 to make it work. I think the feedback of the "sd card presence" pin interfere with the upload.

Sthopeless, 2019-09-10:
@bigtreetech finally I've managed to flash my unit that touchscreen wasnt working and now Is stuck on calibration touch screen and fails everytime i do the red dots (correctly pressed!). what's next?

GenaUser, 2019-09-10:
@Sthopeless Start-up through the calibration screen seems normal after an update (Tested on TFT 3.5 and TFT24).
You need to click on the center of the new point by following these steps :
Click on the red dot in the upper left corner
Click on the red dot in the upper right corner
Click on the red dot in the lower right corner
Click on the black dot in the center of the screen
After, the calibration must be successful and the screen usable.

Sthopeless, 2019-09-10:
@GenaUser I know it is a normal procedure on they're TFT's but this one is constantly failing the calibration, I even tried with touch pen but fails all the time

GenaUser, 2019-09-10:
@Sthopeless I found this information in another topic :
"If you still fail in calibration, Note that the coordinates at the top left of the screen when the get fails, (240,160) are the perfect values, and if it's (0,0), it's a hardware issue of screen.
Please click as far as possible on the center of the point. If possible, sharp objects can be used for calibration to get more accurate values"

blindMoe, 2019-09-10:
When I ran into this issue it seemed to work when using my pinky finger to do the calibration.  It seems odd and pointless but I think it is because of the size of the finger.  So odd... but it worked

Sthopeless, 2019-09-10:
@GenaUser I do get 0,0 at top of the display when it fails, I have also tried countless times including with a touch pen from another 3.5 TFT I have and a small screw driver. Always get "Adjustment failed, please try again."

fabKalix, 2019-09-13:
It is possible to flash over the RS232 port using a USB to Serial TTL adapter if you have one. This does require a wire between BOOT0 and 3v3 which is quite hard to accomplish.
[image] I reverse engineered the two pads for BOOT0 and BOOT1. Both are currently pulled low using R18 for BOOT0 and R5 for BOOT1. If you pull up BOOT0 during power on you can use https://www.st.com/en/development-tools/flasher-stm32.html#get-software to flash the bootloader.

Tungermax, 2019-09-13:
Thanks for the answer, but my knowledge is still not enough to fix the problem. There should be an easier way. It’s not at all clear why sell such a crude product.

oschwieger, 2019-09-14:
> 
> 
> > I have a brand new TFT24 which just arrived.
> > It has v1.1.20 firmware and won't update to v1.1.22
> > It updated the icons and fonts.
> > Can you tell me which filename i need to set? I tried a lot of different ones....
> > Or do i need to update the bootloader first?
> 
> The filename should be: BIGTREE_TFT24_V1.1.22.bin in your case. Placed in the root folder of the SD card. I used Fat32 as the SD card formatting. If it does not work you will probably need the TFT24 bootloader. I would not use the TFT35 posted above. They might be different.

Thanks for pointing me in the Right direction. I used a 64GB SD Card exFat formated. That doesn't seem to work for firmware upgrade. Using a 8GB Fat32 formated card working instantaneous!!!
Anyone having these issue should try using a really small Fat32 formated SD-Card for firmware upgrades.

britmick, 2019-09-16:
TFT32 V2.0 firmware name BIQU_TFT35_APP1_V2.0.22.bin is the one that worked for me without the boot loader update. None of the others above were recognised.

Zuru1, 2019-09-16:
@fabKalix 
Hi, 
You seem to know your way around this display circuit and was wondering if you have found a way to enable the 12864 emulator and also add a rotary encoder?

Thank you

fabKalix, 2019-09-17:
Seems like the variants.h file has a line with:
#define LCD_ENCODER_SUPPORT (defined(LCD_ENCA_PIN) && defined(LCD_ENCB_PIN) && defined(LCD_BTN_PIN))
Above this line there is an if block with only two flavors of display, for the LCD Encoder to work on the TFT3.5 v2.0 we need to add an extra #elif there. Something like this:
//LCD Encoder support
#if defined(TFT24_V1_1)
  //PB0:ENC-A PB1:ENC-B PB2:BTN 
  #define LCD_ENCA_PIN  PB0
  #define LCD_ENCB_PIN  PB1
  #define LCD_BTN_PIN   PB2
#elif defined(TFT35_V3_0)
  //PA8:ENC-A PC9:ENC-B PC8:BTN 
  #define LCD_ENCA_PIN  PA8
  #define LCD_ENCB_PIN  PC9
  #define LCD_BTN_PIN   PC8
#elif defined(TFT35_V2_0)
  #define LCD_ENCA_PIN  PA0
  #define LCD_ENCB_PIN  PA1
  #define LCD_BTN_PIN   PE6   // Use the Leftmost pin of TZ1 as button pin.
#endif

Sthopeless, 2019-09-17:
Bigtreetech showed a video where they disabled the encoder and used the touchscreen onscreen to move around Marlin's display.
I've told them (Bigtreetech) countless times that Marlin's firmware is by far a lot better then they're toucscreen firmware, but they refused :/

fabKalix, 2019-09-17:
Marlin is far better at controlling you printer. I would not dare say it is best at graphics. There is a separation of tasks here. All the display does now is sending GCode commands to Marlin to get information and display it. When printing though I would personally prefer to let Marlin perform the 'heavy work' from SD card instead of depending on the serial connection and the secondary firmware  on the display.

Zuru1, 2019-09-17:
@fabKalix 
Thanks you for point me in the right direction. That last pin was the culprit to solve the encoder. As from my understanding you also mean rightmost pin of TZ1, not leftmost?
Have you tried enabling or running the ST7920 simulator on your TFT35 V2?

fabKalix, 2019-09-17:
> @fabKalix
> Thanks you for point me in the right direction. That last pin was the culprit to solve the encoder. As from my understanding you also mean rightmost pin of TZ1, not leftmost?
> Have you tried enabling or running the ST7920 simulator on your TFT35 V2?

[image]
Yes rightmost, I marked the pin in the image. You could actually use any free pin for the button I think. But this one is closest to the encoder connector.

Zuru1, 2019-09-17:
@fabKalix 
Ok, didn't know that.
Does this also apply to enabling the 12864 emulator? That any of the e.g. USARTx pings can be used to achieve this?

fabKalix, 2019-09-17:
@Zuru1 Yes I think so, According to the stm32f103vc datasheet PB10 and PB11 are USART3_TX and RX. But "Each of the GPIO pins can be configured by software as output (push-pull or open-drain), as
input (with or without pull-up or pull-down) or as peripheral alternate function. Most of the
GPIO pins are shared with digital or analog alternate functions. All GPIOs are high currentcapable."
Not sure if this answers your question. We are getting a bit off topic here by the way

Zuru1, 2019-09-17:
Yeah, I know. But still the only way to get answers. Would have been awesome to get the information from the manufacture, but that ain't happing at the moment. Anyway, I will give it a try and let you know in my issue about this.

bwaldrum, 2019-09-21:
I have this same problem (the **OP** issue of not being able to update the firmware by copying the firmware file to the SD card).  I have the TFT35 V2.0 board and it has the 2.0.20 firmware Like others, I have tried the firmware from this repository and it doesn't upgrade.  I see above that @bigtreetech posted a bootloader, and the only solution I see is to use STLink V2 to update the bootloader, and after that, the documented upgrade process works.  

Questions:
1.  Are all TFT 35 V2.0 boards with the 2.0.20 firmware going to face this problem (essentially un upgradeable)?
2. Will @bigtreetech post an official set of instructions on how to upgrade the bootloader?
3. Will @bigtreetech post the new bootloader in a more "official" and searchable manner?

It seems from the comments that this is a common problem and it would be nice to have a clear, approved set of steps to resolve.

pieterlabie, 2019-09-22:
> It is possible to flash over the RS232 port using a USB to Serial TTL adapter if you have one. This does require a wire between BOOT0 and 3v3 which is quite hard to accomplish.
> [image] I reverse engineered the two pads for BOOT0 and BOOT1. Both are currently pulled low using R18 for BOOT0 and R5 for BOOT1. If you pull up BOOT0 during power on you can use https://www.st.com/en/development-tools/flasher-stm32.html#get-software to flash the bootloader.
> Once initialized, the USART1 (RS232 Port) configuration is: 8 bits, even parity.

Thanks, that worked for me

bullgun, 2019-09-25:
Same issue here the touch screen TFT35 V2.0 doesnt update :-(

bullgun, 2019-09-25:
> 
> 
> Succes! After uploading the new bootloader it worked!

How do you have uploading the bootloader? Thanks

bullgun, 2019-09-25:
> 
> 
> > @EnnajimiYassine I used the boot loader first and flashed it with stink utility
> > [image]
> > [image]
> > [image]
> > And then I just used the precompiled file: BIQU_TFT35_APP1_V2.0.22.bin
> > Thanks a lot for sharing. My TFT35-V2.0 had the same issue.
> 
> I did the same using a this little thing: https://www.aliexpress.com/item/32988024318.html
> Make sure you have the wiring correctly attached to the SWD port. See https://github.com/bigtreetech/BIGTREETECH-TFT35-V2.0/blob/master/TFT3.5-V2.0%20manual.pdf for the pinout of the port.

Its possible do this with arduino nano?

hutchinsane, 2019-09-25:
> It is possible to flash over the RS232 port using a USB to Serial TTL adapter if you have one. This does require a wire between BOOT0 and 3v3 which is quite hard to accomplish.
> [image] I reverse [...]

This worked perfectly for me! Thank you very much for this! Display now updates without a problem :)

bullgun, 2019-09-26:
> This worked perfectly for me! Thank you very much for this! Display now updates without a problem :)

Hi.
Thanks for your help, yes i have a usb to ttl
Do you have a rs232 pinout of the screen for making the connection?
Thanks in advance

fabKalix, 2019-09-26:
> Hi.
> Thanks for your help, yes i have a usb to ttl
> Do you have a rs232 pinout of the screen for making the connection?
> Thanks in advance
The picture below is from bigtreetechs website and shows all pinouts including the r232 port.
[image]

bullgun, 2019-09-26:
> 
> 
> > > > It is possible to flash over the RS232 port using a USB to Serial TTL adapter if you have one. This does require a wire between BOOT0 and 3v3 which is quite hard to accomplish.
> > > > [image] I reverse engineered the two pads for BOOT0 and BOOT1. Both are currently pulled low using R18 for BOOT0 and R5 for BOOT1. If you pull up BOOT0 during power on you can use https://www.st.com/en/development-tools/flasher-stm32.html#get-software to flash the bootloader.
> > > > Once initialized, the USART1 (RS232 Port) configuration is: 8 bits, even parity.
> > > 
> > > 
> > > This worked perfectly for me! Thank you very much for this! Display now updates without a problem :)
> > 
> > 
> > Hi.
> > Thanks for your help, yes i have a usb to ttl
> > Do you have a rs232 pinout of the screen for making the connection?
> > Thanks in advance
> > The picture below is from bigtreetechs website and shows all pinouts including the r232 port. Make sure to connect RX to TX and vice versa. ;)
> > [image]

One more time Thanks
After i made the connection with the ttl, what i should use to upload de firmware? Any software?
Thanks in advance

fabKalix, 2019-09-26:
> One more time Thanks
> After i made the connection with the ttl, what i should use to upload de firmware? Any software?
> Thanks in advance

Make a connection between BOOT0 (marked in an earlier image) and 5V before powering the tft pcb.   This will enter UART bootloader mode. Then use this tool https://www.st.com/en/development-tools/flasher-stm32.html#get-software to flash this image: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/3549545/BIQU_TFT35_V2.0_bootloader.zip Most of this I already posted above in other words. Hope this sums it up so you understand it.
Note this bootloader is only for the TFT3.5 V2.0.

bullgun, 2019-09-27:
> Make a connection between BOOT0 (marked in an earlier image) and 5V before powering the tft pcb. This will enter UART bootloader mode. Then use this tool [...]

i dont understand the first conection BOOT0, i have to put a wire with 5v on that resistor? Here is the 5v? its the BOOT1, I have to conect that two resistors?
I have to have that conection when i power up the tft? Or just connect and disconect and power up? Or conect leave and power up?
What i have to connect on BOOT1? 5V?
Thanks for your pacience, i hope i can do this

fabKalix, 2019-09-27:
> i dont understand the first conection BOOT0, i have to put a wire with 5v on that resistor? Here is the 5v? its the BOOT1, I have to conect that two resistors?
> I have to have that conection when i [...]

Connect the marked soldering pad with a 3v3 pin (I said 5v before, but 3v3 is enough). You can use the first pin from SWD for instance, as you can see that should be 3v3. Keep this connected while booting and flashing. If you are doing it right the stm32 flasher tool will see the device and show it's size otherwise nothing happens. No need to change BOOT1, it is already in the correct state for flashing by default.

bullgun, 2019-09-27:
I Will use USB to TTL , i need 3v3 from somethimg else and the boot1 dont
connect nothing?

fabKalix, 2019-09-27:
> […](#)

I am not sure if the usb to ttl is able to supply the display board. Only connect the rx tx gnd from the usb-ttl to the board and us the usb connector to feed the display board.

bullgun, 2019-09-27:
> 
> 
> > […](#)
> 
> I am not sure if the usb to ttl is able to supply the display board. Only connect the rx tx gnd from the usb-ttl to the board and us the usb connector to feed the display board.

I put a wire with 3v3 on BOOT0 and startup de screen and i just having this. 
[image]

Thanks one more time

[61 more replies]

closed 2021-03-10 by stale[bot]