bigtreetech/BIGTREETECH-TouchScreenFirmware #27: 180-degree rotation of the screen

Acenotass, 2019-08-19:
I was able to achieve screen rotation in the following way:
LCD_Init.c
  LCD_WR_REG(0x36); 
  LCD_WR_DATA(0xE8); //0x28
But this does not affect the touchscreen.
Really need a 180-degree rotation of the screen!!

bigtreetech, 2019-08-28:
Adding this, we intend to add a button in the Settings and click to switch the direction of UI display.
Thank you for your advice.

bigtreetech, 2019-09-09:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/076e808bbd9ff1fc3c34342bb61a7b97e7a47f18

Acenotass, 2019-09-09:
Thank you very much! But I believe that it was not worth it to rotate the screen into the settings, sacrificing the shutdown function. I think it should have been made an option in the configuration file (Configuration.h). In general, it makes sense to remove the configuration file separately from the firmware itself.

oldman4U, 2019-11-02:
Hi.

The way the screen rotation is implemented is useless, because it is temporary as it is lost after turning OFF/ON the printer.

Please implement this feature in a usable way.

Thank you

oldman4U, 2019-11-09:
This seems to work properly with FW 24 on a TFT28 and the new theme is MUCH MUCH better than the themes we had before. I could not test everything, but for me this ticket can be closed.

Thank you BTT

Georg

guruathwal, 2019-11-22:
@Acenotass please close the issue if your problem was solved.

closed 2019-11-24 by Acenotass