bigtreetech/BIGTREETECH-TouchScreenFirmware #31: No display for M117 messages or nozzle position

mickeymicks, 2019-08-25:
Does the display (TFT24 v1.1) support displaying message that were sent via g-code M117? 
Also, the nozzle position. 

The lcd 12864 would have these information (right at it's main page) which i think is very important. However, in the Touchscreen TFT mode, i can't seem to see those information. 

I've created a fork and trying to display the message in a custom menu (page) but i can't seem to find anything in code for the Touchscreen side. The 12864 emulator might have it but it would be better if the touchscreen side also have it by default. 

If there isn't any support for this, could anyone point me in the right direction? 
So, i can add the feature myself. 
e.g. How to receive/parse M117 received from an SKR 1.3 board

(note this is my first time modifying an LCD firmware, so i'm really new to this)

guruathwal, 2019-11-22:
@mickeymicks Status Screen has been added through PR #153 
Please close the issue if it answers you questions.

guruathwal, 2020-01-19:
@bigtreetech  please close this issue.

closed 2020-03-14 by bigtreetech