bigtreetech/BIGTREETECH-TouchScreenFirmware #34: TFT 2.8 firmware do not work

Battou2501, 2019-08-27:
Tried to install firmware on my BigTreeTech BIQU TFT2.8 and getting white screen with crosses in coreners. The only firmware that works is https://github.com/bigtreetech/BIGTREETECH-TFT2.8 but it has no source and for source redirects here.

GenaUser, 2019-09-07:
I have the same problem with the TFT24-V1.1 (BIGTREE_TFT24_V1.1.22.bin)
At starting I have onely the "touch screen calibration".
It's not possible to get out of this screen.

oldman4U, 2019-09-20:
Hello BIGTREETECH. It is great, that you develop new firmware. Unfortunately this does not work with the TFT 2.8, so please remove it from here or even better - please fix it. I have the same problem like the two other users before.

oldman4U, 2019-11-02:
Hello Battou2501, GenaUser.

Seems that the firmware v24 does work now. Could you please confirm, so we can close this ticket.

After uploading the firmware, make sure to press the red and the black calibration points and to add the TFT28 folder with the font and the bmp if you only see destroyed images on screen.

Two issues left for me:
1. The rotate screen function is completely useless, because it is temporary and therefore lost after the next start.
2. Currently I am not able to change the BMPs.

But the firmware upload after compiling using VSC works for me.

Kind regards

GenaUser, 2019-11-02:
Hello @oldman4U 
I think you can close this ticket.
I tested this function on TFT28 TFT35-V1 and TFT35-V3.
That seems work now normally.

Kind regards,

oldman4U, 2019-11-03:
Hi GenaUser.

Great to hear that the new version also works for you. Unfortunately I did not start the ticket and therefore, I can not close it, but Battou2501.

Hi Battou2501.

Could you please check the new firmware and close this ticket if the reported issue is also fixed for you.

Georg

oldman4U, 2019-11-09:
Is there an administrator who could close this ticket. The problem described here is clearly solved and Battou2501 seems not to be active in this forum any more.

Thank you

closed 2019-11-09 by Battou2501