bigtreetech/BIGTREETECH-TouchScreenFirmware #35: How to operate 12864 mode with skr mini e3 board

seksek123, 2019-08-30:
How could I use 12864 mode with SKR mini e3 board?
I can see 12864 mode with press control switch long time but no screen at all.
Touch mode is ok but no screen for 128664 mode.
Please let me know how to use 128664 mode with skr mini e3 board.

thisiskeithb, 2019-09-04:
You can't since there's only one EXP connector on the SKR Mini E3.

dmiart, 2019-09-06:
> You can't since there's only one EXP connector on the SKR Mini E3.

It is EXP3 isn't it? So in theory LCD should still work with the correct pin rearrangement. No SDcard though.

Sthopeless, 2019-09-21:
I have tested TFT35 v3.0 on SKR Mini E3 DIP using EXP3 only and it worked normally.

myudshlihe, 2019-09-27:
> 
> 
> I have tested TFT35 v3.0 on SKR Mini E3 DIP using EXP3 only and it worked normally.

What firmware you used? I can't get 12864 mode with Marlin 2 firmware. It work ok with bigtreetech firmware, but if I build Marlin 2 from marlin repository 12864 mode didn't work...

myudshlihe, 2019-09-27:
I find it!
ENCODER_PULSES_PER_STEP by default was 3 in config example. It need to be changed to 4 and it work.

closed 2020-03-14 by bigtreetech