bigtreetech/BIGTREETECH-TouchScreenFirmware #40: [Request] option to disable sd card in touchscreen mode

spilz87, 2019-09-07:
I really love the TFT2.4 V1.1 screen. 
But I often have issue when I run a print from SD card in Touchmode, absolutely no issue in 12864 emulator mode. 

The touchmode is very convenient to prepare the printer, move axis, etc. 

So I would like to find a way to switch the mode without affecting the printer. 

I think the first step is to be able to disable to the SD card by the SMT32. 

Some one as an idea how to disable it in source code ?

Thanks for your help

Mistergrow, 2019-09-09:
In Marlin you can choose if you want to use the Onboard or LCD SD Card, maybe this will help...?

spilz87, 2019-09-09:
Thanks for your help, 

It can be a way, but it’s not exactly what I want.

I still want to use the SD card slot in lcd board (which is the only one accessible) but I would like only marlin access to it, not the touchscreen processor 

Marlin “alone” works very well, running a print through touchscreen has often issue and stop working. 

I guess the best way is to use the touchscreen to access to functions quickly (move, temperature, home, etc.) and run the print from marlin directly or through G-code like M23 and M24 sent by touchscreen 

Not sure to be clair ^^

closed 2020-08-14 by bigtreetech