bigtreetech/BIGTREETECH-TouchScreenFirmware #41: Changing mode affect the printer

spilz87, 2019-09-09:
The way to change the mode (touchscreen / lcd12864) by holding and turning the rotary encoder is a bad idea : 

Rotary encoder is directly connected to printer, so when we change the mode, we go into menu in printer without knowing it, and sometime that change configuration or something else. 

An other approche could be : 
- touchscreen -> lcd12864 : clic on a specific icon in settings for exemple 
- lcd12864 -> touchscreen : hold a long press on the touchscreen as it’s not used in this mode 

What do you think about that ?

Regards

Mistergrow, 2019-09-09:
I think this is better:
touchscreen -> lcd : touch Settings / Advanced / switch to LCD
lcd -> touch : Go to settings / Advanced / switch to TFT (with encoder)

spilz87, 2019-09-09:
I don’t understand the way lcd -> touch : it suppose you add menu in marlin ?
The point for me is to don’t use encoder which is share with marlin

Mistergrow, 2019-09-09:
Yes, my Idea was Marlin Menu and switch it from there but the unused touchscreen idea from you is a better way.

spilz87, 2019-09-09:
One of my goals is to don’t touch to marlin

guruathwal, 2020-02-20:
@spilz87 can you close this issue.. I think the required option have been added.

closed 2020-03-14 by bigtreetech