bigtreetech/BIGTREETECH-TouchScreenFirmware #43: SKR1.3 + TFT35 v2.0 + WiFi

grego1981, 2019-09-09:
Hi,

I have received my order of an SKR 1.3 with 2209 v1.3 drivers along with a TFT v2.0 and a WiFi expansion from Aliexpress. I'm searching through the repository and internet in general to find some instructions as to make this combination functional, especially for the screen and the WiFi interface. Can someone help me or point me to the right direction?

Thank and best regards.

Gregoris

Mistergrow, 2019-09-09:
Wifi is currently not implemented yet. Bigtree Member said they will publish new source (or Wifi Firmware) Tuesday or Wednesday. (10 or 11.09.2019).

grego1981, 2019-09-09:
So it is going to be implemented on this release (If you have the inside info that is)?

Mistergrow, 2019-09-09:
Look here: 
https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/issues/2#issuecomment-529096010

grego1981, 2019-09-09:
Let's hope it will! The screen has a lot of potential if it can be used to its full extent!

bigtreetech, 2019-09-10:
we will implement it this week. ^_^

sergeyfk05, 2019-09-14:
@bigtreetech , are you implement it(WiFi support) yet?

grego1981, 2019-10-18:
No news still, please give a timeframe. Thank you

closed 2020-08-14 by bigtreetech