bigtreetech/BIGTREETECH-TouchScreenFirmware #52: multi-serial port function (how to connect esp3d and tft to skr 1.3?)

macasero, 2019-09-20:
- multi-serial port function (Touch screen can be directly controlled by ESP3D)

From lates build, my question is where should i connect both devices.

Could I connect ESP3d to aux1 and tftv1.2 to TFT connector.

That way both could share the serial connection?

Thanks

bigtreetech, 2019-09-27:
hello
 - multi-serial port function (Touch screen can be directly controlled by ESP3D) 

this function is only supported on TFT35_V2.0/V3.0
Insert ESP3D directly into TFT35_V2.0/V3.0. just like this
[image]

Actually, AUX-1 and TFT connector are the same port. on SRK-V1.3，So can't use both devices on AUX-1 and TFT connector at the same time.

macasero, 2019-09-27:
That´s what I though. I have tft and esp3d powered with a vcc switch so I can change from one to another without much effort

closed 2020-08-14 by bigtreetech