bigtreetech/BIGTREETECH-TouchScreenFirmware #54: switching mode desable com port receiver

spilz87, 2019-09-21:
tested on TFT35 V3.0 v23

to reproduce : 
- start in TFT mode
- go to temperature menu
- change hotend's temperature
- check you see temperature increase
 -> TFT serial com port receiver works well
- switch to LCD12864 mode
- change bed's temperature 
- switch back to TFT mode
- go to temperature menu
- températures don't change
 -> TFT serial com port receiver doesn't work anymore

do you have the same issue ?

spilz87, 2019-09-21:
Sending data from TFT to Printer seems to work

spilz87, 2019-09-22:
After tests, I think : 
in file "Mode.c"
in function void Serial_ReSourceInit(void)

> Serial_DMAReEnable(SERIAL_PORT);

 is missing

so when go back to TFT mode, in 

> void USART_IRQHandler(uint8_t port)

dma_mem_buf is no more feeled, so 

> infoHost.rx_ok[port]

 never become TRUE again, and void 

> parseACK(void)

 does nothing.

Do I understand well @bigtreetech ?

guruathwal, 2020-02-20:
@spilz87 can you close the issue if it was solved?

closed 2020-03-14 by bigtreetech