bigtreetech/BIGTREETECH-TouchScreenFirmware #55: LCD overview main screen plans?

vMeph, 2019-09-23:
the touch screen seems to be wonderful, but the firmware at this point it still needs alot of development
is there any plans to have a section or a main view where you see everything that is happening with machinewhile is printing?
like the reprap FULL GRAPHIC SMART CONTROLER that shows on main page
the movements of machine while printing
bed temperature actual and target
hotend temperature actual an target
fan percentage
print status
etc....

spilz87, 2019-09-23:
Actually you can’t compare full graphic which is part of Marlin (and so doesn’t need any communication and has no processor) and a separate screen which do lot of more and which has its own Each information you want to have on touchscreen mode has to be send through serial port and in « real time »

extesy, 2019-09-23:
As an example, Wanhao 3d printers use Marlin as their firmware and have colored touch screens which do show real-time information during the print - temperatures, estimates, etc. They don't publish their source code, but what @vMeph is asking is obviously possible.

spilz87, 2019-09-23:
I never had to work on a Wanhao 3d but I guess it’s like for creality cr10s pro : 
They use marlin base but they add some code to have a reliable communication with touch screen processor. 
-> it needs to modify marlin so it’s not compatible here (maybe an idea could be to ask marlin team to have like a standard for this)
-> as it’s a lot of modification, it very difficult to update marlin ...

But if you find a way to do it simple, I’m looking for it to :)

jorgeassuncao, 2019-09-23:
I'm with @vMeph, an overview page like on other graphical screens is a must for this one.

extesy, 2019-09-24:
@bigtreetech sometimes sends pull requests to Marlin codebase so we can ask them too.

extesy, 2019-09-24:
I have created a feature request for Marlin team: https://github.com/MarlinFirmware/Marlin/issues/15348

jessegrunert, 2019-10-03:
+1 on this!

guruathwal, 2020-02-20:
@vMeph should this issue be closed? I think the printing status menu has been added.

closed 2020-03-02 by vMeph

vashadow, 2020-04-15:
Something like M155 - Temperature Auto-Report. There could be an M156 - Status Auto-Report type command added to Marlin so the screens can monitor the status of print jobs with ease. 
M155 S2 (S = seconds)
S:1 F0:255 F1:50 FR:50 X;120 Y:30 Z:23 (S=status 1 for printing, F0=fan0 F1=fan1, etc...)

Just an idea. These newer screens really need a feed from Marlin to be truly useful.