bigtreetech/BIGTREETECH-TouchScreenFirmware #57: TFT24: compile error: RAM overflowed by 3376 bytes

john4242, 2019-09-24:
I cannot even compile it. Got it from github, changed default_envs = BIGTREE_TFT24_V1_1
from platform.io / visual studio code. And I just get an error:
```
Linking .pio\build\BIGTREE_TFT24_V1_1\BIGTREE_TFT24_V1.1.23.elf
c:/users/john/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio\build\BIGTREE_TFT24_V1_1\BIGTREE_TFT24_V1.1.23.elf section `._user_heap_stack' will not fit in region `RAM'
c:/users/john/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: region `RAM' overflowed by 3376 bytes
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\BIGTREE_TFT24_V1_1\BIGTREE_TFT24_V1.1.23.elf] Error 1
```

bigtreetech, 2019-09-27:
sorry for that, have fixed, This array takes up too much RAM
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/351af18746c2e8e2c45b9670a0973146b40b9da4

guruathwal, 2020-02-20:
@john4242  please close the issue.

closed 2020-03-14 by bigtreetech