bigtreetech/BIGTREETECH-TouchScreenFirmware #59: Background Colour (RGB / HEX) / Image + Change Fonts

Underp4ntz, 2019-09-30:
How can we change the Background Colour with RGB or # Colours? or even with a BMP Image?

And there is no information about the FONT and how to change it and how to create a new font and add / replace it...

Can we please get some more information about this stuff? The Thunder has a GUI with images and more information on it and it uses a TFT35?

If you want to sell more of your kits to the DIY community go add more guides / tutorials how to create custom GUI's and FONT's

dineshvyas, 2019-10-07:
It will be really nice to publish GUI documentation.

dineshvyas, 2019-10-10:
i want to add a page. any suggestion?

dineshvyas, 2019-10-15:
I made a new GUI layout. smaller icons and a status bar at bottom to display printer status and kinematics.

thisiskeithb, 2019-10-30:
> I made a new GUI layout. smaller icons and a status bar at bottom to display printer status and kinematics.

Do you have a link?

closed 2020-08-14 by bigtreetech