bigtreetech/BIGTREETECH-TouchScreenFirmware #60: Charset encoding language file.

kkaleta, 2019-09-30:
[image]
Every time I edit a language file, I have a problem with the character encoding.  For example, a degree symbol.  How should these files be encoded?

dineshvyas, 2019-10-10:
recompile and flash it should work or see if it shows something like this in source file.
[image]
you may change these to simple text. for me recompiling and flashing worked.

spilz87, 2019-10-11:
I have the same issue and tried several way but when I change something in langage file in an other #define, °C become Chinese characters :(

spilz87, 2019-10-11:
Where did you find this skin ? I like the icons :)

kkaleta, 2019-10-11:
@spilz87 I've made them.

spilz87, 2019-10-11:
@kkaleta good job 👍

closed 2020-03-14 by bigtreetech