bigtreetech/BIGTREETECH-TouchScreenFirmware #61: connecting TFT 35 V1.2 to Raspberry

Heikoho, 2019-10-01:
Hello,
how can I connect the TFT 35 V1.2 to a Raspberry PI as an normal TFT?
Thank you for answering

Heikoho

drphil3d, 2019-10-04:
Buy the correct hardware

Heikoho, 2019-10-04:
Hello,
what a clever and nice answer.
I like People, who put on there brain before answering.
thanks 
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/61?email_source=notifications\u0026email_token=ANLLSCI3S5FFCZ5NYG474Y3QM2H57A5CNFSM4I4EUZF2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEAKA2CQ#issuecomment-538184970",
"url": "https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/61?email_source=notifications\u0026email_token=ANLLSCI3S5FFCZ5NYG474Y3QM2H57A5CNFSM4I4EUZF2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEAKA2CQ#issuecomment-538184970",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

MangaValk, 2019-10-05:
you can't, wrong hardware

drphil3d, 2019-10-08:
that's all I was saying. I didn't mean to come off as being rude.

bagobor, 2019-10-25:
@Heikoho it's not possible due to lack of 'normal' TFT interfaces. LCD screen is controlled by the onboard STM32 controller.
But you may create your own firmware and use UART interface .... it's definitely not a 'normal' TFT but can be a good and cheap touch controller.  :)

closed 2020-08-14 by bigtreetech