bigtreetech/BIGTREETECH-TouchScreenFirmware #62: French language

GenaUser, 2019-10-01:
Could you add the French language please.
Thanks in advance.

Juju-Dudu, 2019-10-21:
Hello man !

J'ai essayé et ça ne sert pas à grand chose si tu veux tout savoir...

La plupart du vocabulaire qu'on utilise est en anglais et pour avoir essayé, ça met des symboles chinois à la place de certains caractères spéciaux comme le degré...

Du coup, pas utile, mais si tu veux, tu peux télécharger les fichiers sources et les modifier pour créer un nouveau profil de langue... (Pour ma part, j'ai copié le profil english que j'ai traduit, ensuite, j'ai ajouté quelques lignes dans le fichiers de configuration des langues)

En revanche, pour être sûr que ça fonctionne, veille bien à ce que ton écran se mette à jour quand tu mets le firmware (nouveau) sur la racine de la carte sd, sinon, il ne se passera rien...

Vérifie sur le github comment " débloquer " le problème. Moi j'y suis arrivé en passant par le port SWD (écran v2.0).

Bon courage !

guruathwal, 2019-11-10:
The French language was added in PR #74 
Please close the issue.

closed 2019-11-10 by GenaUser