bigtreetech/BIGTREETECH-TouchScreenFirmware #63: Small bug and request

GenaUser, 2019-10-03:
Firmware BIGTREE_TFT35_V3.0.23 (master)
In the Settings menu, the "Rotate UI" has the "Shut down" picture, and "Runout" haven't picture.

Request : Please, could you add Z position in the "Printing interface", and all positions in the "Move interface"

Thanks for your job.

MangaValk, 2019-10-05:
did you copy the images from the repository to the sd card?
it sounds like you didn't add the new images.

GenaUser, 2019-10-05:
@MangaValk you are right.
I didn't think about loading the images separately from the firmware.
Thank you.
The shutdown button disappeared in the new update?

MangaValk, 2019-10-05:
@GenaUser no problem :)
the exchanged the shutdown for the rotate screen button if i remember correctly.

GenaUser, 2019-10-05:
That not easy to use the shutdown module now...
Thanks,

MangaValk, 2019-10-05:
@GenaUser why do you want to shutdown the module? if turns off when you turn off the printer.

GenaUser, 2019-10-05:
I may be wrong
I would like to turn on and shutdown the printer with the "automatic shutdown module" and M80/M81 Gcode
I thought the shutdown button was for this function.

MangaValk, 2019-10-05:
@GenaUser hmm, i dont think it was working for that.

Powerprobot, 2019-10-06:
for update the bitmaps correctly, you must make a folder TFT35 in the root from the SD-card. In this Folder copy the Folder "fonts" and "bmp".

guruathwal, 2019-11-10:
There has been many updates to the firmware since this issue was started. If you problem is solved please close the issue.

closed 2019-11-10 by GenaUser