bigtreetech/BIGTREETECH-TouchScreenFirmware #64: Updating Firmware on TFT35 v2.0.20

mpallen509, 2019-10-05:
I am posting a follow up issue. There was a failure to resolve the issue in the thread: " [Unable to update firmware](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/24) "

There seems to be an inherit issue with TFT35 v2.0 boards prepared with version 20. 

I am unable to update the firmware using the SD card and .bin file names. Multiple names have been suggested and used. Multiple SD cards and formats have also been used.

The best "fix" found in the previous thread was to use an ST-Link device and burn a new bootloader. 

Many others have reported similar issues with the same board and firmware version. 

I am hoping that @bigtreetech can elude to more details and provide a simpler fix for those that purchased the board expecting compatibility and simple firmware editing.

dineshvyas, 2019-10-07:
ST-Link is the easiest solution.

cobrettee, 2019-10-09:
I just got my board today and have this issue so clearly "QC passed" means nothing otherwise it would have been caught and updated before shipping out since it was bought directly from BIQU as well. Now I have to spend more time and money to fix a "new" product.

REDxFROG, 2019-10-11:
What is a USB-2-TTL converter and how to use it?

I am close to just throw this newly arrived TFT right into the garbage.
But then again, I don't even know what the new firmware does. I'm not very happy with 2.0.20.

Using it on a SKR 1.3 (CR-10). The original display has more options and better functionality than this! Sadly have to rate it on Amazon.

cobrettee, 2019-10-12:
@REDxFROG the stlink is better - with the usb-2-ttl you have to connect a wire at teh right time to enter bootloader mode
Amazon:

https://www.amazon.com/DAOKI-ST-Link-Programming-Emulator-Downloader/dp/B01EE4WAC8/ref=sr_1_9?keywords=stlink&qid=1570845052&refresh=1&sr=8-9

Aliexpress
https://www.aliexpress.com/item/32988024318.html?spm=a2g0o.productlist.0.0.5d994e930MrJWz&algo_pvid=f254262a-f719-43f7-a426-e9d534ae88d6&algo_expid=f254262a-f719-43f7-a426-e9d534ae88d6-0&btsid=5af2b39e-6a65-405c-bcbe-2dd5df15b53a&ws_ab_test=searchweb0_0,searchweb201602_4,searchweb201603_52

mavrick01, 2019-10-27:
@bigtreetech  can you confirm what filename you need for V2.0.20? It is very frustrating that we cannot upgrade from v2.0.20.  It seems like some people have been able to move from v2.0.20 to v2.0.22, if this is the case, can you publish v2.0.22 so we can do the steps to 2.0.23 or 2.0.24.1.

HappyCodingRobot, 2019-10-27:
@mavrick01 the bootloader for V2.0 has a hard coded prefix of 'BIQU_TFT35_APP1' for the firmware file it is looking for. If you can not upgrade with theses filenames, you are missing the boot loader. Had this on 2 boards.

mavrick01, 2019-10-27:
Apologies for the ignorance, but how would I use the boot loader? I tried putting it with the BIQU_TFT35_APP1_V2.0.23.bin file, I also tried it with a renames file BIQU_TFT35_APP1.bin and tried  BIQU_TFT35_V2.0_bootloader.bin on its own. Is there a trick to getting this to work?

HappyCodingRobot, 2019-10-28:
The name BIQU_TFT35_APP1_V2.0.xx.bin is fine for the application software. There seems to be a batch of board that has no bootloader flashed, only the application sw. With these you cannot update via sd card. Please see the diskussion mpallen509 referenced on top for a solution to this.

closed 2020-08-14 by bigtreetech