bigtreetech/BIGTREETECH-TouchScreenFirmware #65: Second extruder and second Hot-End not displayed

evandene, 2019-10-08:
My SKR Pro V1.1 is setup with two extruders and two hot ends but the second extruder neither the second hot-end do show up in the menu. (There are also no BMP ICONS for extruder 2 and hot-end 2)
If anyone could help me that would be great.
[image]

evandene, 2019-10-10:
Help is need from @bigtreetech 
TFT35 2.0 is useless for printers with dual extruders / dual nozzles.
The touchscreen doesn't show any second tool or extruder

dineshvyas, 2019-10-10:
it is worrking. try 
[image]
number of icons may be fixed. i made new icon and uploaded but it messed up whole menu.

evandene, 2019-10-10:
It is not working. See my picture in my comment above.
The touch screen does not allow me to operate the second extruder neither the second heater.

evandene, 2019-10-12:
It's not possible to toggle to the second heater neither extruder. #bigtree is not capable for helping.
Is there someone who can help? See my pictures above.

dineshvyas, 2019-10-12:
> It's not possible to toggle to the second heater , fans neither extruder with the existing firmware 2.0.23.
> @bigtreetech is not capable or not willing to help.
> Is there someone who can help? See my pictures above.

Would you like to compile my respiratory and try?

evandene, 2019-10-12:
yeah would be great

dineshvyas, 2019-10-13:
[BIGTREETECH-TouchScreen.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/3721531/BIGTREETECH-TouchScreen.zip)
compiled firmware name will be "BIQU_TFT35_APP1_V2.0.24.bin" I changed it to keep track of my changes.

evandene, 2019-10-13:
Thanks for the firmware.
Next Tuesday, around noon I will be ready with my tests and will inform you

evandene, 2019-10-15:
Test results beta firmware V2.0.24
Firmware virsion not updated in info screen
Only 1 Extruder is shown. Toggling to other extruder is not possible
Only 1 fan is shown.  Toggling to other fan is not possible
Only 1 Heater is shown.  Toggling to other heater is not possible
I stopped the test because of the results. see pictures and movie
Youtube link: https://youtu.be/h8i0jG3dw4k

[image]
[image]

dineshvyas, 2019-10-15:
[BIQU_TFT35_APP1_V2.0.24.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/3729261/BIQU_TFT35_APP1_V2.0.24.zip)
it should work 100%

evandene, 2019-10-15:
New test results
Sorry but I can only come up with exact the same results.
Not even the info screen is okay
I will not test anymore and will scrap the board

evandene, 2019-10-15:
I close the comment

dineshvyas, 2019-10-15:
I think firmware is not updated. you need to flash your TFT35 with bootloader with ST-Link.
this is the bootloader.
[BIQU_TFT35_V2.0_bootloader.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/3730813/BIQU_TFT35_V2.0_bootloader.zip)
This is the only working solution.
i am doing a video will upload within a day. stay tuned don't loose hope.

evandene, 2019-10-22:
Do you have instructions how I need to flash the boot loader? I have a ST-link adapter

oschwieger, 2019-10-22:
try to update the TFT using a really small SD Card. Like 2GB.
This worked for me, i updated 3.5 V2.0 and V3.0 and 2.4 V1.1 using an old SD Card that small and had immediate success on updating the firmware. Make sure its FAT formated.

dineshvyas, 2019-10-22:
https://www.st.com/content/ccc/resource/technical/software/utility/51/c4/6a/b0/e2/0f/47/e5/stsw-link004.zip/files/stsw-link004.zip/_jcr_content/translations/en.stsw-link004.zip
download st-link utility and connect TFT35 v2.0 as in photographs
[image]
[image]
[image]
install st-link utility and click Target
[image]
click connect
[image]
then
Again click target then click Erase Chip
click Ok
click flie 
open file
browse BIQU_TFT35_V2.0_bootloader.zip
click Traget
then Click Program & Verify
do not change anything on prompt and click Start
within few second bootloader will be flashed.
[image]
and your TFT35 V2.0 screen will look like this.
[image]
then flash firmware by SDCARD.
it should work 100%

guruathwal, 2019-11-06:
The TFT35 v2 does not update through sd card.
It can only be updated through flashing via serial TTL or StLink utility. If you don not have STLINK you can flash firmware through Serial with any USB to TTL Converter using STM32 FLASHER here: https://www.st.com/en/development-tools/flasher-stm32.html
Put the board in bootloader mode and then flash the firmware at address: 8003000

evandene, 2019-11-14:
Done

carlray5522, 2020-02-16:
> My TFT35 v2.0 doesn't show second extruder neither second Hot-End
> My SKR Pro V1.1 is setup with two extruders and two hot ends but the second extruder neither the second hot-end do show up in the menu. (There are also no BMP ICONS for extruder 2 and hot-end 2)
> If anyone could help me that would be great.
> [image]

Hello can you please tell me if you ever got the tft35 v2.0 working with dual extrusion?

Charly333, 2020-06-11:
got the same issue with GTR and M5 board. Won´t swap extruders nor fans. but does accept updates, for many new features could be implemented and uploaded.

Anyone solved the problem meanwhile?

closed 2020-08-14 by bigtreetech

przepka1985, 2020-11-09:
> [BIGTREETECH-TouchScreen.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/3721531/BIGTREETECH-TouchScreen.zip)
> compiled firmware name will be "BIQU_TFT35_APP1_V2.0.24.bin" I changed it to keep track of my changes.
> i am trying to add ICON amd LABEL for 2nd Extruder.

Hi. Can i get one for 1 mixing extruder with 3 steppers?