bigtreetech/BIGTREETECH-TouchScreenFirmware #67: TFT35 V3.0 unable to update firmware (MKS-TFT)

akulingus, 2019-10-12:
Hi Everyone,

I recently bought a TFT35 V3.0.  It has the STM32 uC.  I was able to flash the latest TFT35 firmware.   I am using this to interface with a MKS GEN 1.4 controller, so I would like to flash the latest MKS TFT firmware.  For some reason, I can load any compatible version of TFT35 V3.0 firmware, but the MKS firmware won't upload.

I noticed other threads mention renaming the .bin file name.  I intuitively thought I should rename the MKS TFT bin to match the stock firmware bin, but that didn't work.  Anything else I should try?

Thanks, - A

hapklaar, 2019-11-01:
I would think the MKS firmware is not at all compatible with the Bigtreetech TFT35. Have you learned otherwise?

guruathwal, 2019-11-10:
The mks tft has different pin configuration. The firmware wont work.

closed 2020-08-14 by bigtreetech