bigtreetech/BIGTREETECH-TouchScreenFirmware #68: tft35 v3.0 touchscreen "cant find printer) connections problems ENDER 5 v1.1.5

Walstad83, 2019-10-14:
Hi, im not sure how to connect the tft35 v.3 BIGQ to my printer.
My ender 5 with creality silence v.1.1.5 won`t fint the printer when i turn on the touch mode (wich was why i bought it) 

How can i fix it? Any tips is more than welcome and i really apprecitate it

Greetings Daniel

arminth, 2019-11-03:
Try different BAUD rate in display settings. Mine is set to 250000 but default is 115000 so I have to change this after a firmwareupdate.

arminth, 2019-11-03:
Add: Where on the 1.1.5 did you connect the serial cable!?

arminth, 2019-11-03:
You probably didn't connect serial cable. This is needed to use Touchscreen. This page from ESP3D might help you:
https://github.com/luc-github/ESP3D/wiki/Creality-CR-10-Ender-3

closed 2020-08-14 by bigtreetech