bigtreetech/BIGTREETECH-TouchScreenFirmware #75: How to change or add font

mooner1022, 2019-10-18:
Is there any manual about changing or adding font files for any other languages? I'm trying to add korean to the firmware, but the font doesn't print korean.

dmiart, 2019-10-20:
A while ago someone posted links to modified fonts for Cyrillic, but this program can probably be used for any bitmap type as it's universal.  

http://corosan.ru/sgfed/

https://github.com/Acenotass/BIGTREETECH-TFT35-V1.2/tree/master/firmware/Copy%20to%20SD%20Card%20root%20directory%20to%20update/TFT35/font

mooner1022, 2019-10-22:
> A while ago someone posted links to modified fonts for Cyrillic, but this program can probably be used for any bitmap type as it's universal.
>
> http://corosan.ru/sgfed/ [...]

Thx for the reply, but I wasn't able to open files with this program.

thisiskeithb, 2019-10-30:
> Thx for the reply, but I wasn't able to open files with this program.

The app works and can edit fonts in a Windows XP VM, but I also found that it couldn’t read any of the TFT font files.

dmiart, 2019-10-30:
https://github.com/bigtreetech/BIGTREETECH-TFT35-V1.2/issues/41

This is the original post. I assumed it worked for creating fonts for the display.  There is an export function in this program and there is way to export as binary.

guruathwal, 2019-11-08:
You can check out word_unicode.fon file. It contains the fonts for oher languages. You may check it in SG Font editor.  Try these instructions
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/103#issuecomment-550180539
You will need to change doc properties size 24x24 and total count to a very high number to view all the characters.

mooner1022, 2019-11-09:
I'll try it when I get back from the trip. Thx guys!

mooner1022, 2019-11-09:
> You can check out word_unicode.fon file. It contains the fonts for oher languages. You may check it in SG Font editor. Try these instructions
> [#103 (comment)](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/103#issuecomment-550180539)
> You will need to change doc properties size 24x24 and total count to a very high number to view all the characters.

I've applied the fonts and firmware, but korean still does not show and white boxes are showing instead. At least it isn't cracking like the original font, so I think it's showing some hope.

guruathwal, 2019-11-09:
> I've applied the fonts and firmware, but korean still does not show and white boxes are showing instead. At least it isn't cracking like the original font, so I think it's showing some hope.

What Screen size do you have?
**for TFT35**
you need to import the `word_unicode.fon` with these settings
- Doc Properties
[image]

  Symbols count: 8192 ( this is the maximum count allowed in SG font Editor )

- Import Setting
  [image]

**For TFT24/28**
- Doc Properties
  - Change Symbol width & height to 16

- Import Setting
  - Change Byte per line 2

Export Setting will be same as import setting.

you need to check the unicode character address of the language you want to add.

mooner1022, 2019-11-09:
> What Screen size do you have?
> **for TFT35**
> you need to import the `word_unicode.fon` with these settings
>
> * Doc Properties
>   [image]
>   Symbols count: 8192 ( this is the maximum count allowed in SG [...]

The screen I have is TFT35 V2. I'll try to modify the file, thanks!

mooner1022, 2019-11-09:
> What Screen size do you have?
> **for TFT35**
> you need to import the `word_unicode.fon` with these settings
>
> * Doc Properties
>   [image]
>   Symbols count: 8192 ( this is the maximum count allowed in SG [...]

I'm trying to edit the file, but I'm facing a problem. Korean decimal starts from 12593, but the bitmap editor's limit is 8192. How can I solve this?

guruathwal, 2019-11-11:
> I'm trying to edit the file, but I'm facing a problem. Korean decimal starts from 12593, but the bitmap editor's limit is 8192. How can I solve this?

till now we did not find any free software to work over that.
also can't confirm if the TFT supports characters over 8192, this will require deep digging into the code, or if @bigtreetech could confirm that it would be nice.

mooner1022, 2019-11-11:
> till now we did not find any free software to work over that.
> also can't confirm if the TFT supports characters over 8192, this will require deep digging into the code, or if @bigtreetech could [...]

Then is there any way to add language on unused spaces in font file and specify the position in firmware to read it korean?

closed 2020-08-14 by bigtreetech