bigtreetech/BIGTREETECH-TouchScreenFirmware #81: [BUG][TFT24] Several bad prints at the same point

bagobor, 2019-10-25:
Several bad prints at the same point

- was printer from onboard SD card
- several attempts with power-cycling all boards in between 
- looks like a gcode handling issue by onboard controller
- the same file was printed just fine by using OctoPrint.

jimmyeao, 2019-10-29:
Same issue here using the TFT SD Card. Print stops at end of first layer with heaters left on. Reset required to regain control. FW V3.0.24.1

bagobor, 2019-10-29:
@jimmyeao Sounds like it's very similar to my case. Initially, I thought I was related to Marlin's buffer overflow issues in some debug log function but Marlin's update didn't change anything. 
I'll check for my FW version.

bagobor, 2019-12-09:
@jimmyeao it looks like the latest firmware update is addressing this issue.  But it introduces new memory issues....

jimmyeao, 2019-12-09:
> @jimmyeao it looks like the latest firmware update is addressing this issue. But it introduces new memory issues....

What issues are you seeing?

bagobor, 2019-12-11:
@jimmyeao example: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/0d4b3f99873f25d21ed9c0c0fd609e9a89c94970/TFT/src/User/API/Gcode/gcode.c#L78

(the function returns a pointer to deallocated (clearRequestCommandInfo();) mem buffer.)

closed 2020-08-14 by bigtreetech