bigtreetech/BIGTREETECH-TouchScreenFirmware #83: Compatability

oldman4U, 2019-10-29:
Hi.

According to BTT, this firmware should be compatible with the following boards:
TFT35 V1.0/V1.1/V1.2/V2.0/V3.0, TFT28, TFT24 V1.1

My question is if there is a single user out there using a TFT28 successfully with this firmware and IF yes, if he could provide us the files he is using or a firmware. I bought the TFT28 because it was mentioned to be compatible and i saw that there is a lot of development going on, but all i get is a white screen using this firmware.

If this firmware is not compatible i would like to see BTT to make this officially and remove the screen from the list of compatible devices.

Thank you

thisiskeithb, 2019-10-30:
The TFT35 2.0 isn’t updatable initially without an ST-Link, so this LCD might have a similar issue: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/24

bigtreetech, 2019-10-30:
hello TFT28 is supported in here, you should upload this [BIQU_TFT28_V1.0.23.bin](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/BIQU_TFT28_V1.0.23.bin), and this [TFT28](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/TFT28) folder. I just tested and confirmed that the firmware in this link worked normally

oldman4U, 2019-10-30:
Dear bigtreetech.

Using the .bin works, but you describe the ability to use platformio to compile the firmware to add or change features, get rid of bugs and so on. How can this be achieved using the TFT28.

Please let me know

Thank you

oldman4U, 2019-10-30:
Hello thisiskeithb.

Thank you for your answer. I saw this also, but it is not that i can not upload the firmware. The problem is, that I can not compile a new version following the description provided but that i have to stay with an older version. This BIGTREETECH-TouchScreenFirmware should support the TFT28 but it is not possible to compile it so the TFT28 can use it. For me this is at least confusing if not a fraud.

thisiskeithb, 2019-10-30:
The firmware compiles fine for me with `default_envs` set to `BIGTREE_TFT28`:

```prolog
Linking .pio/build/BIGTREE_TFT28/BIQU_TFT28_V1.0.24.1.elf
Building .pio/build/BIGTREE_TFT28/BIQU_TFT28_V1.0.24.1.bin
Checking size .pio/build/BIGTREE_TFT28/BIQU_TFT28_V1.0.24.1.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=====     ]  51.0% (used 25044 bytes from 49152 bytes)
PROGRAM: [===       ]  27.3% (used 71636 bytes from 262144 bytes)
======================================= [SUCCESS] Took 6.93 seconds =======================================

Environment         Status    Duration
------------------  --------  ------------
BIGTREE_TFT35_V1_0  IGNORED
BIGTREE_TFT35_V1_1  IGNORED
BIGTREE_TFT35_V1_2  IGNORED
BIGTREE_TFT35_V2_0  IGNORED
BIGTREE_TFT35_V3_0  IGNORED
BIGTREE_TFT28       SUCCESS   00:00:06.931
BIGTREE_TFT24_V1_1  IGNORED
======================================= 1 succeeded in 00:00:06.931 =======================================
```

oldman4U, 2019-10-30:
Hi thisiskeithb.

This is perfect and the expected behaviour. So far this also works for me and finally makes me believe it should work. But do you have a TFT28 and have you tried to upload the firmware and to use the screen after that. I can also upload the firmware on my screen, but after that all i get is a white screen with calibration points. And this is the case since i got the screen. But i have to say that i have not tested the very last version, but the one older than 24h.

Maybe i am just not smart enough to make this work, but i would be happy to learn from another user using this screen successfully following the instructions provided by BBT.

thisiskeithb, 2019-10-30:
> But do you have a TFT28

I only have the BigTreeTech TFT24 1.1 and TFT35 3.0, sorry.

oldman4U, 2019-10-30:
;-) Thought so.

But thank you very much for your help.

Regards

Georg

thisiskeithb, 2019-10-30:
Does the attached .bin work on your TFT28? This was just compiled.

[BIQU_TFT28_V1.0.24.1.bin.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/3789077/BIQU_TFT28_V1.0.24.1.bin.zip)

oldman4U, 2019-10-30:
[image]

Hello thisiskeithb.

This happens when I upload the .bin you sent me 7h ago.

;-(

bigtreetech, 2019-11-01:
hello, you need to click the new red dots (three in total) and then the last black dot in the center of the screen to calibrate the screen, After that upload this [TFT28](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/TFT28) folder.  This folder contains normal font and icon files, and the screen should work after this

oldman4U, 2019-11-02:
Dear BTT.

IT WORKS, THANK YOU.

Because I have tried doing the same many many times before, I guess this has been fixed with the latest firmware.

I close this ticket, even I have still two issues.

Two issues left for me:
1. The rotate screen function is completely useless, because it is temporary and therefore lost after the next start.
2. Currently I am not able to change the BMPs.

But I will do more tests and open a new ticket if I can not fix this.

Again, thank you for your help. I am happy to be able to close this ticket.

Georg

closed 2019-11-02 by oldman4U