bigtreetech/BIGTREETECH-TouchScreenFirmware #86: Font Change?

jhnnyn, 2019-10-30:
I've been trying to change the font on my screen. I can't figure out what type of file the font is stored as. The readme says it's a bitmap font, but I've tried several converters and viewers for bitmap and raster fonts but can't seem to open them. Is there a special program used to view/edit the fonts?

thisiskeithb, 2019-10-31:
Take a look at https://github.com/bigtreetech/BIGTREETECH-TFT35-V1.2/issues/41, specifically [this comment](https://github.com/bigtreetech/BIGTREETECH-TFT35-V1.2/issues/41#issuecomment-508440690) as it has a link to font templates you can open with SG Bitmap Font editor.

thisiskeithb, 2019-11-02:
FYI: Some things have changed in recent commits related to font handling, so the fonts above no longer work. Hopefully it'll get resolved soon!

closed 2020-07-04 by bigtreetech