bigtreetech/BIGTREETECH-TouchScreenFirmware #88: Please add a status screen

hapklaar, 2019-11-01:
To make this screen a little more useful, can you add a status screen that shows while the printer is printing something? I would like to see the details you normally see on an LCD12864 all together on this status screen:

- Printing time
- Progress (when printing from SD)
- Nozzle temperature 
- Bed temperature
- X/Y/Z position
- Fan speed
- Info that can be placed using M117

JRoz17, 2019-11-01:
This is a must for these screens to be a complete upgrade from the LCD12864

arminth, 2019-11-03:
I second that!! Without, I am just using the 12864 emulation as I am printing from octoprint.

Patag, 2019-11-19:
Just pull the latest commits ! It's a giant step for users with this new status screen. This new UI is quite perfect 👍 . I'd just like a small change in fan speed display : a % value would be more user friendly than a 0-255 raw value. Except this, I want to congratulate the devs. Keep on this great work.
Thanks

guruathwal, 2019-11-21:
@hapklaar  you can close this now... 👍

hapklaar, 2019-11-21:
@guruathwal right :)

closed 2019-11-21 by hapklaar

JRoz17, 2019-11-22:
> Just pull the latest commits ! It's a giant step for  with this new status screen. This new UI is quite perfect  . I'd just like a small change in fan speed display : a % value would be more user friendly than a 0-255 raw value. Except this, I want to congratulate the devs. Keep on this great work.
> Thanks @Msq001

Could you point me to the right direction? Github is not too intuitive for me and I just keep getting lost. I have tried searching for tft35 v2 but does not seem to take me to any update for the screen.

Really appreciate any help you could provide.

UnknownHero99, 2019-11-22:
> Could you point me to the right direction? Github is not too intuitive for me and I just keep getting lost. I have tried searching for tft35 v2 but does not seem to take me to any update for the [...]

download code that's available on github and compile it using platformio