bigtreetech/BIGTREETECH-TouchScreenFirmware #90: Latest version does not compile for TFT24

davehardy20, 2019-11-02:
I saw the changes to the report this morning and 'pulled' the changes and tried to compile for TFT24 v1.1, but got these errors

> Executing task in folder BIGTREETECH-TouchScreenFirmware: C:\Users\dave\.platformio\penv\Scripts\platformio.exe run <

Processing BIGTREE_TFT24_V1_1 (platform: ststm32; framework: stm32cube; board: STM32F105RC)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/STM32F105RC.html     
PLATFORM: ST STM32 5.7.0 > STM32F105RC (48k RAM. 256k Flash)
HARDWARE: STM32F105RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-stm32cube 2.0.181130
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <BSP-Common>
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\API\Gcode\gcode.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\API\Gcode\mygcodefs.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\API\LCD_Encoder.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\API\Language\Language.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\API\Language\utf8_decode.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\API\UI\GUI.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\API\UI\ST7920_Simulator.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\API\UI\touch_process.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\API\UI\ui_draw.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\API\Vfs\vfs.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\API\boot.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\API\extend.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\API\interfaceCmd.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\API\parseACK.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Fatfs\myfatfs.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Hal\LCD_Init.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Hal\stm32f10x\Serial.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Hal\stm32f10x\lcd.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Hal\xpt2046.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\BabyStep.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\Extrude.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\Fan.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\Heat.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\Home.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\MainPage.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\Mode.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\Move.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\Popup.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\PowerFailed.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\Print.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\Printing.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\ProbeOffset.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\Settings.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\Speed.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\leveling.o
TFT\src\User\Menu\Mode.c: In function 'menuMode':
TFT\src\User\Menu\Mode.c:92:11: warning: unused variable 'nowEncoder' [-Wunused-variable]
   int16_t nowEncoder = encoderPosition = 0;
           ^~~~~~~~~~
TFT\src\User\Menu\Print.c: In function 'menuSendGcode':
TFT\src\User\Menu\Print.c:240:90: warning: pointer targets in passing argument 5 of 'GUI_DispStringInRect' differ in signedness [-Wpointer-sign]
     GUI_DispStringInRect(LCD_WIDTH-GCODE_KEYW-START_GKX,10,LCD_WIDTH-10,10+BYTE_HEIGHT+5,"Send", 0);
                                                                                          ^~~~~~
In file included from TFT\src\User/includes.h:22:0,
                 from TFT\src\User\Menu\Print.c:2:
TFT\src\User\API\UI/GUI.h:55:6: note: expected 'const uint8_t * {aka const unsigned char *}' but argument is of type 'char *'
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\menu.o
 void GUI_DispStringInRect(int16_t sx, int16_t sy, int16_t ex, int16_t ey, const uint8_t *p, uint8_t mode);
      ^~~~~~~~~~~~~~~~~~~~
TFT\src\User\Menu\Print.c:241:70: warning: pointer targets in passing argument 5 of 'GUI_DispStringInRect' differ in signedness [-Wpointer-sign]
     GUI_DispStringInRect(10,10,GCODE_KEYW+START_GKX,10+BYTE_HEIGHT+5,"Back", 0);
                                                                      ^~~~~~
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\main.o
In file included from TFT\src\User/includes.h:22:0,
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\os_timer.o
                 from TFT\src\User\Menu\Print.c:2:
TFT\src\User\API\UI/GUI.h:55:6: note: expected 'const uint8_t * {aka const unsigned char *}' but argument is of type 'char *'
 void GUI_DispStringInRect(int16_t sx, int16_t sy, int16_t ex, int16_t ey, const uint8_t *p, uint8_t mode);
      ^~~~~~~~~~~~~~~~~~~~
TFT\src\User\Menu\Settings.c:200:12: warning: 'item_runout_i' defined but not used [-Wunused-variable]
 static u8  item_runout_i = 0;
            ^~~~~~~~~~~~~
Linking .pio\build\BIGTREE_TFT24_V1_1\BIGTREE_TFT24_V1.1.24.1.elf
.pio/build/BIGTREE_TFT24_V1_1/src/src/User/API/interfaceCmd.o: In function `sendQueueCmd':
c:\Users\dave\3D-Printing\BTT_TFT_Firmware\BIGTREETECH-TouchScreenFirmware/TFT\src\User\API/interfaceCmd.c:230: undefined reference to `positionSetUpdateWaiting'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\BIGTREE_TFT24_V1_1\BIGTREE_TFT24_V1.1.24.1.elf] Error 1
======================================================================================== [FAILED] Took 4.22 seconds ========================================================================================

Environment         Status    Duration
------------------  --------  ------------
BIGTREE_TFT35_V1_0  IGNORED
BIGTREE_TFT35_V1_1  IGNORED
BIGTREE_TFT35_V1_2  IGNORED
BIGTREE_TFT35_V2_0  IGNORED
BIGTREE_TFT35_V3_0  IGNORED
BIGTREE_TFT28       IGNORED
BIGTREE_TFT24_V1_1  FAILED    00:00:04.218
================================================================================== 1 failed, 0 succeeded in 00:00:04.218 ================================================================================== 
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

Seems as thought there is problems with the code.

hapklaar, 2019-11-02:
Also compile errors with TFT35, maybe the pull request from other forks were not checked?

[image]

UnknownHero99, 2019-11-02:
i believe this occurs only on tft's with encoder right? it works fine on my TFT35 V2.0
i believe the is because not properly implementing buzzer disable/enable feature
on TFT with encoder both buzzer and LCD12864 buttons have same KEY_ICON_ 
and then it returns switch case duplicate error

UnknownHero99, 2019-11-02:
TFT35 V3 compiles ok if u change  
line 34  `#define LCD12864_BG_INDEX KEY_ICON_3` in ScreenSettings.c 
to `#define LCD12864_BG_INDEX KEY_ICON_4`

UnknownHero99, 2019-11-02:
For TFT35 V3 change
ScreenSettings.c
line 32
`#ifdef ST7920_SPI // LCD12864 color settings
  #ifdef BUZZER_PIN
    #define LCD12864_BG_INDEX KEY_ICON_3
  #else
    #define LCD12864_BG_INDEX KEY_ICON_4
  #endif
  #define LCD12864_FN_INDEX (LCD12864_BG_INDEX+1)`

to `#ifdef ST7920_SPI // LCD12864 color settings
  #ifdef BUZZER_PIN
    #define LCD12864_BG_INDEX KEY_ICON_4
  #else
    #define LCD12864_BG_INDEX KEY_ICON_3
  #endif
  #define LCD12864_FN_INDEX (LCD12864_BG_INDEX+1)`

hapklaar, 2019-11-02:
That fixed it, thanks!

UnknownHero99, 2019-11-02:
no problem, but unfortunately this still doesnt work for TFT24 1.1

spilz87, 2019-11-02:
in extend.c file, move 

> /* Set whether we need to query the current position */
void positionSetUpdateWaiting(bool isWaiting)
{
  update_waiting = isWaiting;
}

outside of the #ifdef FIL_RUNOUT_PIN

UnknownHero99, 2019-11-02:
made pull request working version avaliable at https://github.com/UnknownHero99/BIGTREETECH-TouchScreenFirmware/tree/TFT35V3fix works for TFT24 V1.1 too thanks to @spilz87

arminth, 2019-11-06:
Issue solved with today's commit.

guruathwal, 2019-11-08:
@davehardy20 please confirm and close the issue.

closed 2019-11-08 by davehardy20