bigtreetech/BIGTREETECH-TouchScreenFirmware #94: French staggered labels (last November 02 update)

GenaUser, 2019-11-02:
First, Thanks for added French translation and the compatible font.

There is a little bug in the "fr_pack Label" list. Some button names are staggered.
You find below the modification for this pack.

I added too, a "language_fr.c" file proposition in agreement with the 10 characters limitation, and with accented letters.

/Language.c

const char *const fr_pack[LABEL_NUM]={
  FR_LANGUAGE,
  FR_HEAT,
  FR_MOVE,
  FR_HOME,
  FR_PRINT,
  FR_EXTRUDE,
  FR_FAN,
  FR_SETTINGS,
  FR_LEVELING,
  FR_POINT_1,
  FR_POINT_2,
  FR_POINT_3,
  FR_POINT_4,
  FR_ABL,
  FR_BLTOUCH_TEST,
  FR_BLTOUCH_DEPLOY,
  FR_BLTOUCH_STOW,
  FR_BLTOUCH_REPEAT,
  FR_PROBE_OFFSET,
  FR_EEPROM_SAVE,
  FR_INC,
  FR_DEC,
  FR_NOZZLE,
  FR_BED,
  FR_STOP,
  FR_BACK,
  FR_PAGE_UP,
  FR_PAGE_DOWN,
  FR_PAUSE,
  FR_RESUME,
  FR_LOAD,
  FR_UNLOAD,
  FR_SLOW_SPEED,
  FR_NORMAL_SPEED,
  FR_FAST_SPEED,
  FR_FAN_FULL_SPEED,
  FR_FAN_HALF_SPEED,
  FR_ROTATE_UI,
  FR_TOUCHSCREFR_ADJUST,
  FR_MORESETTING,
  FR_SCREFR_INFO,
  FR_WHITE,
  FR_BLACK,
  FR_BLUE,
  FR_RED,
  FR_GREEN,
  FR_CYAN,
  FR_YELLOW,
  FR_BROWN,
  FR_GRAY,
  FR_DISCONNECT,
  FR_BAUDRATE_115200,
  FR_BAUDRATE_250000,
  FR_RUNOUT_OFF,
  FR_RUNOUT_ON,
  FR_SMART_RUNOUT_ON,
  FR_PERCENTAGE,
  FR_BABYSTEP,
  FR_PERCENTAGE_SPEED,
  FR_PERCENTAGE_FLOW,
  FR_VALUE_ZERO,
  FR_1_DEGREE,
  FR_5_DEGREE,
  FR_10_DEGREE,
  FR_X_INC,
  FR_Y_INC,
  FR_Z_INC,
  FR_X_DEC,
  FR_Y_DEC,
  FR_Z_DEC,
  FR_X_HOME,
  FR_Y_HOME,
  FR_Z_HOME,
  FR_001_MM,
  FR_01_MM,
  FR_1_MM,
  FR_5_MM,
  FR_10_MM,
  FR_1_PERCENT,
  FR_5_PERCENT,
  FR_10_PERCENT,

  FR_READY,
  FR_BUSY,
  FR_UNCONNECTED,
  FR_DISCONNECT_INFO,
  FR_LOADING,
  FR_POWER_FAILED,
  FR_CONTINUE,
  FR_CANNEL,
  FR_ADJUST_TITLE,
  FR_ADJUST_INFO,
  FR_ADJUST_OK,
  FR_ADJUST_FAILED,
  FR_WARNING,
  FR_STOP_PRINT,
  FR_CONFIRM,
  FR_TFTSD,
  FR_READ_TFTSD_ERROR,
  FR_TFTSD_INSERTED,
  FR_TFTSD_REMOVED,
  FR_U_DISK,
  FR_READ_U_DISK_ERROR,
  FR_U_DISK_INSERTED,
  FR_U_DISK_REMOVED,
  FR_ONBOARDSD,
  FR_READ_ONBOARDSD_ERROR,
  FR_FILAMENT_RUNOUT
};

/Language_fr.h
#ifndef _LANGUAGE_FR_H_
#define _LANGUAGE_FR_H_

    #define FR_LANGUAGE             "Français"
    #define FR_HEAT                 "Chauffer"
    #define FR_MOVE                 "Déplacer"
    #define FR_HOME                 "Origines"
    #define FR_PRINT                "Impression"
    #define FR_EXTRUDE              "Extruder"
    #define FR_FAN                  "Ventiler"
    #define FR_SETTINGS             "Paramètres"
    #define FR_LEVELING             "Niveaux"
    #define FR_POINT_1              "Point 1"
    #define FR_POINT_2              "Point 2"
    #define FR_POINT_3              "Point 3"
    #define FR_POINT_4              "Point 4"
    #define FR_ABL                  "ABL"
    #define FR_BLTOUCH_TEST         "Test"
    #define FR_BLTOUCH_DEPLOY       "Deploy"
    #define FR_BLTOUCH_STOW         "Stow"
    #define FR_BLTOUCH_REPEAT       "Repeat"
    #define FR_PROBE_OFFSET         "Offset"
    #define FR_EEPROM_SAVE          "Sauver"
    #define FR_INC                  "Plus"
    #define FR_DEC                  "Moins"
    #define FR_NOZZLE               "Tête"
    #define FR_BED                  "Plateau"
    #define FR_STOP                 "Stop"
    #define FR_BACK                 "Retour"
    #define FR_PAGE_UP              "Précédent"
    #define FR_PAGE_DOWN            "Suivant"
    #define FR_PAUSE                "Pause"
    #define FR_RESUME               "Reprise"
    #define FR_LOAD                 "Extruder"
    #define FR_UNLOAD               "Retrait"
    #define FR_SLOW_SPEED           "Lent"
    #define FR_NORMAL_SPEED         "Normal"
    #define FR_FAST_SPEED           "Rapide"
    #define FR_FAN_FULL_SPEED       "Max"
    #define FR_FAN_HALF_SPEED       "Demi"
    #define FR_ROTATE_UI            "Tourner UI"
    #define FR_TOUCHSCREFR_ADJUST   "Ajust TS"
    #define FR_MORESETTING          "Autres"
    #define FR_SCREFR_INFO          "Info"
    #define FR_WHITE                "blanc"
    #define FR_BLACK                "noir"
    #define FR_BLUE                 "bleu"
    #define FR_RED                  "rouge"
    #define FR_GREEN                "vert"
    #define FR_CYAN                 "cyan"
    #define FR_YELLOW               "jaune"
    #define FR_BROWN                "marron"
    #define FR_GRAY                 "gris"
    #define FR_DISCONNECT           "Déco"
    #define FR_BAUDRATE_115200      "115200"
    #define FR_BAUDRATE_250000      "250000"
    #define FR_RUNOUT_OFF           "Détect OFF"
    #define FR_RUNOUT_ON            "Détect ON"
    #define FR_SMART_RUNOUT_ON      "Smart ON"
    #define FR_PERCENTAGE           "Avances %"
    #define FR_BABYSTEP             "BabyStep"
    #define FR_PERCENTAGE_SPEED     "Vitesse"
    #define FR_PERCENTAGE_FLOW      "Débit"
    #define FR_VALUE_ZERO           "Zéro"
    #define FR_1_DEGREE             "1℃"
    #define FR_5_DEGREE             "5℃"
    #define FR_10_DEGREE            "10℃"
    #define FR_X_INC                "X+"
    #define FR_Y_INC                "Y+"
    #define FR_Z_INC                "Z+"
    #define FR_X_DEC                "X-"
    #define FR_Y_DEC                "Y-"
    #define FR_Z_DEC                "Z-"
    #define FR_X_HOME               "X"
    #define FR_Y_HOME               "Y"
    #define FR_Z_HOME               "Z"
    #define FR_001_MM               "0.01mm"
    #define FR_01_MM                "0.1mm"
    #define FR_1_MM                 "1mm"
    #define FR_5_MM                 "5mm"
    #define FR_10_MM                "10mm"
    #define FR_1_PERCENT            "1%"
    #define FR_5_PERCENT            "5%"
    #define FR_10_PERCENT           "10%"

    #define FR_READY                "Prêt"
    #define FR_BUSY                 "     Occupé, veuillez attendre..."
    #define FR_UNCONNECTED          "L'imprimante n'est pas reliée!"
    #define FR_DISCONNECT_INFO      "Vous pouvez maintenant piloter par l'ordinateur!"
    #define FR_LOADING              "Chargement..."
    #define FR_POWER_FAILED         "Continuer l'impression?"
    #define FR_CONTINUE             "Continuer"
    #define FR_CANNEL               "Annuler"
    #define FR_ADJUST_TITLE         "Calibration écran tactile"
    #define FR_ADJUST_INFO          "SVP Touchez le point rouge"
    #define FR_ADJUST_OK            "Ajustement OK"
    #define FR_ADJUST_FAILED        "Ajustement non OK, essayez encore"
    #define FR_WARNING              "Attention"
    #define FR_STOP_PRINT           "Arrêter l'impression?"
    #define FR_CONFIRM              "Confirmer"
    
    #define FR_TFTSD                "SD TFT"
    #define FR_READ_TFTSD_ERROR     "Erreur de lecture carte SD écran!"
    #define FR_TFTSD_INSERTED       "Carte inserée!"
    #define FR_TFTSD_REMOVED        "Carte enlevée!"
    
    #define FR_U_DISK               "Clé USB"
    #define FR_READ_U_DISK_ERROR    "Erreur de lecture clé USB!"
    #define FR_U_DISK_INSERTED      "Clé USB insérée!"
    #define FR_U_DISK_REMOVED       "Clé USB  enlevée!"
    
    #define FR_ONBOARDSD            "SD interne"
    #define FR_READ_ONBOARDSD_ERROR "Erreur de Lecture carte SD interne!"
    
    #define FR_FILAMENT_RUNOUT      "Fin de Filament!"
    
#endif

bigtreetech, 2019-11-04:
very thanks for your help, I have merged it in https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/99

guruathwal, 2019-11-07:
please close the issue if the problem is solved.

closed 2019-11-07 by GenaUser