bigtreetech/BIGTREETECH-TouchScreenFirmware #96: Latest version doesn't compile for TFT35 v1.2 + Fatfs warnings

ArturNadolski, 2019-11-03:
Linking .pio\build\BIGTREE_TFT35_V1_2\BIQU_TFT35_V1.2.24.1.elf
.pio/build/BIGTREE_TFT35_V1_2/src/src/User/API/interfaceCmd.o: In function `sendQueueCmd':
d:\_3D\Soft\TFT35v1.2_Nadol/TFT\src\User\API/interfaceCmd.c:230: undefined reference to `positionSetUpdateWaiting'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\BIGTREE_TFT35_V1_2\BIQU_TFT35_V1.2.24.1.elf] Error 1

There is a bunch of warnings at Fatfs.
Example:
In file included from TFT\src\User\Fatfs\ff.h:29:0,
                 from TFT\src\User\Fatfs\ff.c:22:
TFT\src\User\Fatfs\ffconf.h:171:34: warning: excess elements in array initializer
 #define FF_VOLUME_STRS  "SD","U","CF","SD","SD2","USB","USB2","USB3"

UnknownHero99, 2019-11-03:
same as #90 
fix [here](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/92)

ArturNadolski, 2019-11-09:
Fixed with latest releases

closed 2019-11-09 by ArturNadolski