bigtreetech/BIGTREETECH-TouchScreenFirmware #98: Compiling for TFT35 V3.0 Fails

kstanley74, 2019-11-03:
Tried several times to compile and received the following error in the build panel.

Compiling .pio\build\BIGTREE_TFT35_V3_0\src\src\User\delay.o
TFT\src\User\Menu\ScreenSettings.c: In function 'menuScreenSettings':
TFT\src\User\Menu\ScreenSettings.c:140:7: error: duplicate case value
       case LCD12864_BG_INDEX:
       ^~~~
TFT\src\User\Menu\ScreenSettings.c:131:7: note: previously used here

       case BUZZER_KEY_INDEX:
       ^~~~
*** [.pio\build\BIGTREE_TFT35_V3_0\src\src\User\Menu\ScreenSettings.o] Error 1
Compiling .pio\build\BIGTREE_TFT35_V3_0\src\src\User\main.o

UnknownHero99, 2019-11-04:
errors are fixed in latest merges :P

kstanley74, 2019-11-04:
Thank you, Jakob!

Kevin

hapklaar, 2019-11-09:
Please close if it is fixed for you!

kstanley74, 2019-11-09:
Closing. This issue has been resolved.

closed 2019-11-09 by kstanley74