bigtreetech/BIGTREETECH-TouchScreenFirmware #101: Complie errors due bad merging

UnknownHero99, 2019-11-04:
@bigtreetech there were compile errors for 2 merges in a row now please create development branch and merge it with master when tested, check pull requests more carefully and i guess merge only pull requests with no conflicts, or at least after merging check for compile errors before pushing it to master

version before making it [useless](https://github.com/UnknownHero99/BIGTREETECH-TouchScreenFirmware)

bigtreetech, 2019-11-04:
Complie errors has be fixed now. Actually, I'm a novice of GIT  ^_^, so sorry for that. 
missed icons will be uploaded later, (TFT24/TFT28 icons and font also will be uploaded lader)

UnknownHero99, 2019-11-04:
no problem, everyone needs to start somewhere but please consider making development branch :P

closed 2019-11-04 by UnknownHero99

UnknownHero99, 2019-11-04:
@bigtreetech oh and thumbs up because ur applying merges so quickly and bltouch icons are nice too :D