bigtreetech/BIGTREETECH-TouchScreenFirmware #105: Still garbled fonts on TFT24 even after BigtreeTech cleanup and fixed compile commit

arminth, 2019-11-04:
Nothing readable in the menus! It's the same in the standard and in the unified UI.
[image]

davehardy20, 2019-11-04:
Same here too, although the font issue looks different
[image]

UnknownHero99, 2019-11-04:
would you try older commits and help find the commit that made this bug, here on TFT35 V2 it works just fine

arminth, 2019-11-04:
Version 1.1.24 (which I can't find under releases anymore) worked. I downloaded it October 15.

arminth, 2019-11-04:
@davehardy20 
I had tried to copy the 3.5" fonts into the fonts-folder. Since then, I have these strange noisy thingis!
Before, it was identical to yours. Probably, it saved garbage into the memory.

GrishaKh, 2019-11-04:
I also have this problem.

GrishaKh, 2019-11-04:
It seems like everything is fine when you use the ready-made build, but when you build it yourself and download the binary file to the chip, this problem arises.

davehardy20, 2019-11-05:
I think the garbled font issue showed up with pull request #89. Also there is a new pull request #103 which is a possible fix for this issue.

arminth, 2019-11-05:
I doubt it's a fix as there are no changes to the fonts in TFT24 Directory from what I can derive from the PR.

thisiskeithb, 2019-11-05:
@bigtreetech Maybe revert the font-related changes for now? They seem to be more trouble than they're worth at the moment (and we can return to using [much better fonts](https://github.com/bigtreetech/BIGTREETECH-TFT35-V1.2/issues/41)).

arminth, 2019-11-06:
With today's commit I can confirm that garbled font-issue is resolved (speaking of english, german amd I also accodentally had spanish at bootup). I can't verify cyrillic fonts.
I also realized that in fullscreen Marlin-Mode, image is not distorted any more. Height is adapted correctly now leaving some space top and bottom. Looks good!
What I found:

Icons Missing in Preheat Menu: PLA ABS PETG Both
In Screen Menu: Icon is not changing for Screen color - looks a bit stupid when it says "Red" but Icon is still blue. Not a big deal but kind of not logic. Misspell: Jellow instead of Yellow. Colors not translated into German except for blue. RED=ROT GREEN=GRUEN BLUE=BLAU YELLOW=GELB WHITE=WEISS BLACK=SCHWARZ GREY=GRAU BROWN=BRAUN CYAN=CYAN

Button function not clear:
Menu Setup->Machine->Custom ??
Menu Feature->Manual ??

Other German translations: 
FAN=Luefter
Machine=Maschine (or maybe better "Drucker" which means printer as you wouldn't normally say Maschine in German.

Menu Title of the Leveling Menu says ABL instead of Leveling.
Pressing the Offset-Button shows a Message with the probe Z-Offset. Pressing ok, the same message pops up again (tested on Marlin 2.0.x, 15.9.2019). 

Sorry for reporting this way, but I am not capable of correctly using Git. I hope, somebody from @bigtreetech reads this and leverage from it.
For now, I close the issue as my problem is solved.
Thanks
Armin

closed 2019-11-06 by arminth

thisiskeithb, 2019-11-06:
> I also realized that in fullscreen Marlin-mode, image is not distorted any more. Height is adapted correctly now leaving some space top and bottom. Looks good!

I couldn't take it anymore, so I had to fix it 🙂

> In Screen Menu: Icon is not changing for screen color - looks a bit stupid when it says "Red" but Icon is still blue. 

This won't change since it's a [single static bitmap](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/03487f732df427357a5f61b222f9cab88b5df8f6/Copy%20to%20SD%20Card%20root%20directory%20to%20update/TFT35/bmp/BackGroundColor.bmp).

mbribeiro, 2020-04-21:
Same for me, my LCD is now useless, should I throw  it in the trash since this is not fixed ?