bigtreetech/BIGTREETECH-TouchScreenFirmware #111: [BUG] Custom ST7920_BKCOLOR & _FNCOLOR Have No Effect

thisiskeithb, 2019-11-06:
No matter what I define for `ST7920_BKCOLOR` & `ST7920_FNCOLOR`, the custom colors are ignored.

They can be changed in TFT mode, but do not change after firmware updates on two TFT35 V3s that I own.

guruathwal, 2019-11-06:
Did you tried changing the color through the settings menu in the TFT Display?

thisiskeithb, 2019-11-06:
> Did you tried changing the color through the settings menu in the TFT Display?

Sorry if that wasn't clear. That's what I meant by:

> They can be changed in TFT mode

guruathwal, 2019-11-06:
> > Did you tried changing the color through the settings menu in the TFT Display?
> 
> Sorry if that wasn't clear. That's what I meant by:
> 
> > They can be changed in TFT mode

You also mentioned
> They can be changed in TFT mode, but do not change after firmware updates on two TFT35 V3s that I own.
What does "but do not change after firmware updates" means here.
Anyways the color settings are saved in system eeprom / flash storage so that they cam be restored after reset.
When the system finds settings stored in memory then it should load those settings and discard any setting set through configuration.h.
If your colors setting changed through TFT menu survives throught hard reset then it is working fine.

thisiskeithb, 2019-11-06:
> What does "but do not change after firmware updates" means here.

If I change the background color in Configuration.h and flash the firmware, it is not updated/uses previous color settings.

If I change the background color through TFT mode, it works and survives reboots.

guruathwal, 2019-11-06:
> What does "but do not change after firmware updates" means here.
> 
> If I change the background color in Configuration.h and flash the firmware, it is not updated/uses previous color settings.
> If I change the background color through TFT mode, it works and survives reboots.

:thumbsup: This is the normal behaviour which is to be expected. due to the following code:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/03487f732df427357a5f61b222f9cab88b5df8f6/TFT/src/User/main.c#L40-L43

If it does not reads  a valid setting it will pickup settings from the configuration.h file:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/03487f732df427357a5f61b222f9cab88b5df8f6/TFT/src/User/Menu/Settings.c#L7-L20

You can call `infoSettingsReset()` to restore the default colors but it will reset all the settings with it.

You can also write another function to reset only colors like:
```
void ST7920_reset(void)
{
  infoSettings.bg_color = ST7920_BKCOLOR;
  infoSettings.font_color = ST7920_FNCOLOR;
  storePara();  
}
```
then call it from your code.

:v:

thisiskeithb, 2019-11-06:
> This is the normal behaviour which is to be expected.

I thought this worked on an older commit, so I’ll do some more digging.

Thank you for pointing me in the right direction!

thisiskeithb, 2019-11-06:
I reverted to b98cdf796d3f0ef271dd035a873a653bd0547a88 and defining `ST7920_BKCOLOR` & `ST7920_FNCOLOR` in `Configuration.h` works.

 I'll have to dig through the code and figure out why the change was made.

guruathwal, 2019-11-08:
You can close the issue for now. Open it again if there is any new detail.

thisiskeithb, 2019-11-08:
Has it been fixed?

guruathwal, 2019-11-08:
There is nothing to be fixed.
If you want you can add a reset color button in Touch screen settings and push a PR for the same.
Sugestion:
Add code in `boot.c` in the icon update part to reset colors when icons are updated or when a specific file is found in the root directory eg. "Reset_color.txt".

thisiskeithb, 2019-11-08:
> There is nothing to be fixed.

There is if It worked in the past and doesn’t anymore :)

guruathwal, 2019-11-08:
> > There is nothing to be fixed.
> 
> There is if It worked in the past and doesn’t anymore :)

It can be..
What was the last commit where it was working?

thisiskeithb, 2019-11-09:
> It can be..
What was the last commit where it was working? 

[I reverted](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/111#issuecomment-550515886) to b98cdf7 and confirmed it works. When I have more time, I’ll work through commits and find the exact change that broke it.

thisiskeithb, 2020-03-10:
Closing as this seems to be resolved in the last round of updates.

closed 2020-03-10 by thisiskeithb

thisiskeithb, 2020-04-01:
For anyone that may find this thread, I'd recommend creating a blank `reset.txt` file and placing on the SD card to reset all settings since some old settings around through firmware updates.