bigtreetech/BIGTREETECH-TouchScreenFirmware #113: [Question] TFT35 V3.0 - Wifi (ESP01S) implementation

Patag, 2019-11-06:
Is there any functional code for ESP01S module when plugged in the dedicated LCD connector ?
If yes, where can I find documentation/tutorial on set-up ?
Thanks.
BTW, the new Unified theme is great ! Bravo !

carl1961, 2019-11-06:
works with pronterface, https://github.com/luc-github/ESP3D

Patag, 2019-11-07:
Ok, thanks. I already did that.
But I'm trying to set up a SKRPro 1.1 / TFT35 v3/ ESP01S config. I'm currently able to get wifi when module is plugged on SKR but in this case TFT is no more usable (serial port used by ESP). So I'm trying to make ESP functional when plugged in TFT. It looks like ESP doesn't even boot in this config. Any idea ?

guruathwal, 2019-11-08:
Try a simple blink sketch from arduino ide on the esp then plug it in the display and check if it is working or not

kordris, 2019-11-08:
I would also like some more info on this, I gotten the a sketch loaded from Adruino setup on the module, then had it seen on my wifi and even set it up. before I even turned it off. I pluged it on the back of the tft display. I don't have it picking up on my wifi either. So is there anything need to be done on the tft display or what. There is no kinda documentation on this. Come on you guess need to give some more info on how this should be handled.

Patag, 2019-11-08:
Exactly same issue here.

Aleks-MO, 2019-11-08:
https://www.facebook.com/groups/505736576548648/permalink/768393890282914/

huczas, 2019-11-08:
@Aleks-MO 
https://www.facebook.com/groups/505736576548648/permalink/768393890282914/
Can you write here what’s there those who don’t have a Facebook account?

Aleks-MO, 2019-11-08:
https://scontent-arn2-1.xx.fbcdn.net/v/t1.15752-9/71550618_1276625229191403_5913931428403347456_n.jpg?_nc_cat=111&_nc_oc=AQl3q11lQLTYGlxBsnBl9I1yoQxStFXW9IP3CYerGAnnp7iGEFQUKzETWntifn75DoE&_nc_ht=scontent-arn2-1.xx&oh=46b2b291cdcfd94a9002542f48b8c490&oe=5E44CF77
Flip pins for TFT35 v3.0
https://scontent-lga3-1.xx.fbcdn.net/v/t1.0-9/71293326_136541530997653_8271056587649449984_n.jpg?_nc_cat=108&_nc_oc=AQnaAjUi974iAOXKbqp7eC48K3px_rQUZCy832ZGnu-qH4jhMgQAt0hwAScPQ7Te4a4&_nc_ht=scontent-lga3-1.xx&oh=a3dc1f4dfc777068731c4e7d2a9c220d&oe=5E55F559

Set as here.
https://scontent-lga3-1.xx.fbcdn.net/v/t1.0-9/70966260_137029790948827_2585636416033652736_n.jpg?_nc_cat=111&_nc_oc=AQmT2YkOeRq21l_liHMSl2AYsiSAYLCP-WuPMWWDqNM8fn17-_Qb5W4ZkOxIJ7nDgEA&_nc_ht=scontent-lga3-1.xx&oh=74eaa71156cd3818af9f9d5ac840ff3c&oe=5E54D20F

guruathwal, 2019-11-08:
if it was working before connecting to the board then you should check if there is +5V on "ch_pd" pin of ESP after adding it to the board and turn it on.

huczas, 2019-11-08:
As I understand that fist photo "flip pins on esp01" it's completely wrong in comparison with a official pinout
https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/blob/master/BIGTREETECH%20TFT35%20V3.0/Hardware/TFT35%20V3.0-PIN.pdf

kordris, 2019-11-08:
Is there anything on the marlin software side that we need to update to get this working?

Aleks-MO, 2019-11-08:
@huczas Tak, zrozumieli poprawnie, zdjęcie połączenia COM-TTL3.3v i instalację programu do programowania ESP. Sam BIN znajduje się w folderze https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/tree/master/firmware/esp-01s       Przepraszamy za automatyczne tłumaczenie..)))

Aleks-MO, 2019-11-08:
[image]
Tak powinno być.    
[image]

huczas, 2019-11-08:
@Aleks-MO thanks that translation was pretty much ok 😉 lets stay with English.
After burning that bin file on all of my 3 esp01's all of them are showing same in serial:
```
 ets Jan  8 2013,rst cause:4, boot mode:(3,4)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
```
This is my setup for flashing:

[image]

[image]

Do you have any hint?

Aleks-MO, 2019-11-08:
https://sprut.ai/client/blog/950  
Description, it is necessary so. Better without buttons. [https://sprut.ai/static/media/cache/00/00/70/40/950/33117/3000x_image.jpg?1552119510](url)

Firmware program
https://youtu.be/wYaJr2cYFnc?t=162

bin address 
[image]
Everything should work out.

Aleks-MO, 2019-11-08:
After the firmware, ESP became an access point! When connecting for the first time, the address is 192,168,0,1 
Default Configuration
Default Settings:

Access Point: ESP3D
PW:12345678
Authentification: WPA
IP: 192.168.0.1
Baud rate: 115200
Web port:80
Data port: 8888
if Authentication is enabled :
User: admin
Password: admin
User:user
Password: user

Description here.
https://github.com/luc-github/ESP3D

Sthopeless, 2019-11-09:
Is not possible to implement the TFT's flashing the ESP3D bin file from the sdcard to the ESP01?! 
This would help a lot of users, I am seeing every now and then a post on facebook asking how to flash the esp01's with esp3d

guruathwal, 2019-11-09:
the Pins on TFT do not match with Pins of ESP
[image]

Aleks-MO, 2019-11-09:
@Sthopeless  NO!

guruathwal, 2019-11-09:
> [image]
> Tak powinno być.
> [image]

Fliping the ESP will leave the EN / CH_PD pin of esp on NC pin of the TFT. EN / CH_PD needs to be pulled up with 3v for the ESP to work.

Aleks-MO, 2019-11-09:
did not understand your question. BTT just do not stand on the side of the board put the plug, "inside out."

carl1961, 2019-11-09:
As you see in the picture the pins need to be heated up with a soldering iron and shoved through to the other side of the board

guruathwal, 2019-11-09:
> did not understand your question. BTT just do not stand on the side of the board put the plug, "inside out."

a question and a doubt: the ESP is seen working in the pictures above when it is flipped, but then the EN (Enable) pin of the esp connect to the NC (no connection : not connected to any thing) pin on the boards as it is labeled on the board schematics. The EN pin of the ESP need to be connected to 3v for it to work. then how is it working if it is not connected to anything?

Patag, 2019-11-09:
I confirm this pin-out, at least for power supply. Did not apply it as is but with Dupont wires (to stay compatible with my USB adapter/programmer). This ESP module out of the box fits for SKR Pro but needs flip for TFT.

Aleks-MO, 2019-11-09:
@guruathwal  
Everything works and is very good. You just have to try.

kordris, 2019-11-09:
Ok, I got mine working with jumper wires since it's not same as the board. But is there any info on how to update the info on getting the printer info. I can pull up some info but it's not really letting do anything. Its pulling some data under the printer tab.

guruathwal, 2019-11-09:
@Aleks-MO  I have TFT35 V2.

Hukuma1, 2019-11-12:
The confusion is the newest version (TFT35 V3) with the wifi module finally works as intended... just plug it in and it powers on. Prior versions you had to fix by soldering I believe.

Patag, 2019-11-12:
Newest ? Since what date ? I've a TFT35 V3 and need to flip esp01s connector. Is there as V3.x ? ATM, I can't find any confusion !

Hukuma1, 2019-11-12:
[image]

I received it about 3 weeks ago. Ordered from Aliexpress. E3 v1.2 along with wifi module. Different wifi module?

Aleks-MO, 2019-11-12:
@Hukuma1 Show me the work.

Sthopeless, 2019-11-12:
my v3.0 also works with the esp01

Hukuma1, 2019-11-12:
[image]

Ignoring the ugly popup UI, it does work. Although does your wifi module stop working when you issue a M503 command, @Sthopeless? I can issue it, it reports back the info, but then ceases to accept anymore commands unless I reset the TFT. Wondering if bug?

Sthopeless, 2019-11-12:
Sorry I didn't test any further, I saw it connected and that was it as I dont use esp3d. I check now what is flashed on the esp01 atm or if is not esp3d i will test M503 tomorrow

guruathwal, 2019-11-13:
If @Hukuma1 have the same display as others. Then it might be just mislabeled pins.

Patag, 2019-11-13:
[image]
[image]

Esp connector label is different here.
What is your version (as seen on first picture) ?

huczas, 2019-11-15:
In my version the label is "ESP" and I can confirm that it works when I just plug normal esp01. If it't upsidedown it is not powered. 

Second thing in my case is that after boot up my esp01 even if its on - is not connecting and not powering properly, it means that there is no wifi, and no any sign on screen that it's working. 
red light on esp01 is on that's it.

If I power my esp01 to FTDI to my PC and open putty I see that it is booting and showing hided ssid:ESP3D with IP:192.168.4.1 with no password.

Is there a possibility to connect ESP01 to different RX/TX connector?

huczas, 2019-11-15:
Last shot that I want to compare, @Hukuma1 can you show your configuration.h part with serial ports are configured?
In my board, when I use 12v and connect also usb my PC can't connect do the board. The only way to pass Gcode is a USB stick or SD card connected to TFT.

marcocm, 2019-11-22:
For what I have seen, there are two TFT35 V3.0.
One that has a ESP label and other that has a ESP-8266.
If you have the TFT with the ESP label, you can just connect the WIFI ESP module.
 In my case, I can connect the printer and control de printer using ESP3D. 
I just can't use the WEBUI SD options. I can't upload files .gco or .gcode and I can't see the files in the SD Card. I think I might have to update some configuration on marlin/ESP3D/Screen firmware. Also, to use the SD card do I have to connect some EXP1/EXP2/EX3 cable? I am just using the TFT cable to connect the Screen to de BBT SKR V1.3 board.

I read some where that the board and the screen worked bette with the BAUDRATE of 250000, and I tried it and I confirmed it.

Hukuma1, 2019-11-24:
Sorry for the late reply, @huczas! I am using standard Marlin 2 branch settings. This is for SKR Mini E3 v1.2 board:

'/**
 * Select the serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT 2

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 -1'

ORGATHM, 2020-01-20:
> > [image]
> > Tak powinno być.
> > [image]
> 
> Fliping the ESP will leave the EN / CH_PD pin of esp on NC pin of the TFT. EN / CH_PD needs to be pulled up with 3v for the ESP to work.

what is the name of the pronterface style wifi tool?
thanks

Aleks-MO, 2020-01-20:
**@ORGATHM** I did not understand the question. If you are talking about the look in the browser, then read here - https://github.com/luc-github/ESP3D-WEBUI

ORGATHM, 2020-01-20:
@Aleks-MO oo, sorry. i was in the webinterface with my phone but i could see this pronterface style interface.
my fault. i think i have to upload some thing into the small interface.

[image]

Aleks-MO, 2020-01-20:
@ ORGATHM 
Of your files marked in red, take them here  https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/tree/master/firmware/esp-01s
[image]
[image]

SargFowler, 2020-01-28:
Just as a thought, would I actually require a display if I just wired the ESP module up to my SKR V1.3 board serial pins? I could access the printer via the webpage.

Aleks-MO, 2020-01-28:
@SargFowler    It’s more convenient to work with him....

Aardvarx55, 2020-02-26:
Very keen to get the Wifi operational on my just installed BTT-TFT35-E3 V3.0, and although the ESP8266 is powered, it doesn't appear to boot (the blue led seem on continuously, which it doesn't when plugged into my programming rig, where it becomes visible on the network).
Noted earlier comment on board marking i.e. ESP and ESP8266, but mine is 'WIFI" (see photos).
Anyone got one of these 'WIFI' marked boards working?
[image]
[image]

ssombra, 2020-02-26:
Is it possible to connect a wifi module to the TFT35 V2.0 ? It doesn't have a socket but in the manual there is a section labeled wifi with pins labeled: RX, EN, RST, 3.3V, GND, NC, NC, TX

Aardvarx55, 2020-02-26:
It's got a socket for a EAP8266, it just doesn't seem to work....
I think this is the latest board, as it has the extra board below the encoder.

redonebe, 2020-02-29:
I have the same problem. TFT 3.5 V3.0. Both LED's on the ESP keep on. I measured the board and pins EN and RST are not high so the ESP can not work.

Aardvarx55, 2020-03-01:
I've got my ESP8266 working now...
The schematic for the TFT35-E3 V3.0 shows the ch_pd pin as NC (not connected), which I believed would stop the ESP booting, as the ch_pd needs to be high.
So I connected the it via jumper cables, but connected ch_pd to VCC, and it boots whilst connected to the TFT board... result.
Having proved that it would boot, I soldered a link directly on the ESP module, and now I can insert it without the jumper cables, and it works.
I have ESP3D running on it, have it connected to my home network, and am getting responses from the printer... early days for me, as I've never used ESP3D, but my aim is to be able to upload .gcode files via WiFi (I currently print from SD card)
[image]
[image]

Scope666, 2020-03-01:
I wonder if that solder mod is necessary if you're using BTT's ESP-01S board?

Aleks-MO, 2020-03-01:
@Scope666  No.

Scope666, 2020-03-01:
> @Scope666 No, you don't need anything.

Yeah, that's what I figured.  I'm guessing it's because he's using another vendor's board.

Aleks-MO, 2020-03-01:
@Scope666      
There is ESP 01 another ESP01S, they are different .... if you do not dig deep ...))

Aardvarx55, 2020-03-01:
Interesting. I hadn't realised there were two types... Anyway, that's what I have been using for other projects, so at least it's only a minor mod, and just for testing.
Thanks for the heads up.

Aleks-MO, 2020-03-01:
@Aardvarx55   
[image]
[image]

Aardvarx55, 2020-03-01:
I see the schematic has CH_PD connected to VCC :o)

redonebe, 2020-03-01:
@Aardvarx55 

Best practice is to put a 10K resistor between 3,3V and CH_PD and one between 3,3V and reset. I will certainly work with a direct connection but it is safer with a 10K resistor.

This is how I did it.
[image]

Aardvarx55, 2020-03-02:
You are course quite right. I should have used a 10K resistor for the 'pull up' on CH_PD.
The RESET pin is addressed by the TFT board, so I haven't played with that :o)
I'm afraid I just copied what my programming rig has.
Will have to change both...
Thanks for your help, Robbie

[28 more replies]

closed 2020-08-14 by bigtreetech