bigtreetech/BIGTREETECH-TouchScreenFirmware #118: [Suggestion] Different layout for showing filesystem

Hukuma1, 2019-11-07:
[image]

Did a quick mockup of how it currently is, and what I think could be an improvement. Right now it's hard to read the filenames and you have to wait for it to scroll through to confirm you got the right one. This way you would have more room to see the whole file. Any thoughts?

[image]

guruathwal, 2019-11-07:
This would be nice.

Sthopeless, 2019-11-07:
this would be very nice!

Aleks-MO, 2019-11-07:
Well read long file name, more information.

thisiskeithb, 2019-11-11:
This is how file selection works on the MKS Nano (with “dumb” TFT), or at least on the Lotmaxx SC-10:

[image]
Source: [Chris Taylor/Nerys - Lotmaxx SC 10 Analysis T3DP](https://youtu.be/n_JcMkiBRfw)

I like the compactness of the buttons on the right and we’re all in agreement that a list would improve readability.

Aleks-MO, 2019-11-11:
@thisiskeithb  
Better option 2, lines! The icons here are redundant.

guruathwal, 2019-11-11:
I agree with @Aleks-MO

Sthopeless, 2019-11-11:
I also agree with @Aleks-MO & @guruathwal the list looks by far better!

thisiskeithb, 2019-11-11:
Gcode previews do look nice, though:

[image]

Sthopeless, 2019-11-11:
yes they do. maybe isn't possible to have it as a selectable option(?) one way shows the menu vertically where we can read the names properly the other one has the preview mode in horizontal view.

Aleks-MO, 2019-11-11:
@thisiskeithb 
Do you need these little thingies and lace? Need a print! )))

Hukuma1, 2019-11-11:
[image]

How about both?

Aleks-MO, 2019-11-11:
Need to see a photo of the screen?

Hukuma1, 2019-11-11:
[image]

You mean like this?

Aleks-MO, 2019-11-11:
Yes! ))

Hukuma1, 2019-11-18:
Updated the previews of the idea above. Slight change to accommodate for space on screen.

ShaggyDog18, 2019-11-23:
Absolutely support that development: the current Print list with 8-char file names looks... like jumping back to the last century. Please, re-design this page, show full file names in the vertical list... No doubt that would improve usability of TFT screens.

Aleks-MO, 2019-11-23:
@ShaggyDog18  Little willing, everyone wants icons... It came to the point that the vertical location of the screen for its filling with information was accepted.

guruathwal, 2019-11-24:
[image]

Welcome back again to 21st century #179

Hukuma1, 2019-11-24:
[image]

Thank you, @guruathwal. :) I would like to suggest we mask the file name extension (.gcode). Or maybe an option to? And how about replacing device source text (SD:/) with icon as well? 21st century looking nice. :D

Aleks-MO, 2019-11-24:
@guruathwal Did it come in? It is perfectly! 
I have not seen it yet. ¯\_(ツ)_/¯

Sthopeless, 2019-11-24:
thank you @guruathwal looks ALOT! better, but mine is missing up, down, back icons? I see only 2 squares on each button

guruathwal, 2019-11-24:
@Sthopeless you need to update the fonts also..

Hukuma1, 2019-11-24:
@Sthopeless jumped ahead into 23rd century with those missing font alien symbols. :D

guruathwal, 2019-11-24:
> @guruathwal Did it come in? It is perfectly!
> I have not seen it yet. ¯_(ツ)_/¯

👍

Sthopeless, 2019-11-24:
@Hukuma1 true that :P 
I didn't even noticed the folder and files icons were also missing

Aleks-MO, 2019-11-24:
@guruathwal    when they upgrade package to public access  the TFT35 and bin (version xxx.24.3)???  ;-(

Hukuma1, 2019-11-24:
@Aleks-MO I'm impatient too. So I downloaded the files before they were approved for master branch:

Download these 2 files and re-flash your firmware from SD card:

1. https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/b487c6fe9b16626ef57a8dfc4805a699edab5dd1/Copy%20to%20SD%20Card%20root%20directory%20to%20update%20-%20Unified%20Menu%20Material%20theme/BIGTREE_TFT35_V3.0.25.bin

2. https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/b487c6fe9b16626ef57a8dfc4805a699edab5dd1/Copy%20to%20SD%20Card%20root%20directory%20to%20update%20-%20Unified%20Menu%20Material%20theme/TFT35/font/word_unicode.fon

Replace the FON file that's in your TFT35\font directory.

Just tested it on TFT35 V3. Looks great!

Aleks-MO, 2019-11-24:
@Hukuma1  ))) 
Thank you, I will try .. My personal opinion is that everything should be brought to a single system and issue the final version here. Downloaded, installed. Not all TFT programmers!

Hukuma1, 2019-11-24:
I think they need to make material theme the default and simply delete the other icons/theme altogether. ;) And split downloads to specific models for less tech-savvy users. That way TFT35 V3.zip can simply have the 95x95 icons, and the corresponding firmware.bin. So install would be as easy as unzipping to SD card for upgrade. Not searching which folder to choose, etc.

Aleks-MO, 2019-11-24:
Yes,  there will be less confusion ...

guruathwal, 2019-11-25:
@Aleks-MO @Hukuma1 
I will try to simply the files and update process to be easier for everyone..👍

Hukuma1, 2019-12-05:
[image]

Maybe we can have an intermediary screen before we start print? Could be useful if a) you wanted to confirm a print file (visual), or b) maybe you clicked on wrong file because a list view is slightly harder to click than icons.

Aleks-MO, 2019-12-05:
@Hukuma1  My opinion is superfluous ... Better think, why 2 com? Can it be used for LEDs?

Hukuma1, 2019-12-05:
Bad translation? Not sure what you mean?

Aleks-MO, 2019-12-05:
Думаю это лишнее.  Как использовать дополнительные порты COM на TFT.  Можно  его использовать для светодиодов?

Hukuma1, 2019-12-06:
No idea about additional COM ports on the board or anything about LEDs. That's for another topic I think. :P

GenaUser, 2019-12-06:
@Hukuma1 I agree with your visual proposition.

Hukuma1, 2019-12-06:
You’ll have to see if @guruathwal think it’s possible to implement. 👍🏼

Aleks-MO, 2019-12-06:
@Hukuma1  Он (@guruathwal) здесь все управляет?

guruathwal, 2019-12-19:
@Aleks-MO  not at all.. I am just a contributor 👍 .
@Hukuma1 sure it is possible, but need some time.. 👍

Aleks-MO, 2019-12-19:
@guruathwal  Жаль..))   Хозяина здесь не хватает... 
Видел твои обновления , пожалуйста делай фото основного экрана и экрана печати., для большей наглядности как это выглядит. Кто пробовал, говорят что русские шрифты очень плохие.  Спасибо что правишь прошивки. 
@guruathwal Is a pity.)) The master here is missing...
Saw your updates, please take a photo of the main screen and the print screen., for more clarity, what it looks like. Who tried, they say that Russian fonts are very bad. Thanks for ruling the firmware.

closed 2020-03-19 by Hukuma1

gordo3di, 2022-11-08:
I've recently been trying to get the thumbnails to show up again in the printing screen. After two full days of messing around I am no further along. Something is wrong in the ICON_PREVIEW code but I still can't figure out why. I was also trying to get the thumbnail to show up in the start print popup which is what started all this.

In PrintingMenu.c I just forced the model_DirectDisplay to see if I could get it to show up. In the code below it will show for a brief millisecond then is immediatly replaced by the move icon.

```c
    printingItems.items[KEY_ICON_5].icon = (infoFile.source < FS_ONBOARD_MEDIA && isPrintModelIcon()) ? ICON_PREVIEW : ICON_BABYSTEP;
    //gCreate testing to make preview thumb work
    //printingItems.items[KEY_ICON_5].icon = (infoFile.source < FS_ONBOARD_MEDIA && isPrintModelIcon()) ? ICON_BABYSTEP : model_DirectDisplay(getIconStartPoint(5), infoFile.path);
    printingItems.items[KEY_ICON_5].icon = model_DirectDisplay(getIconStartPoint(5), infoFile.path);
```

Changing it to ICON_PREVIEW will have some icon show up that I've never seen with no brief thumbnail.

```c
    printingItems.items[KEY_ICON_5].icon = (infoFile.source < FS_ONBOARD_MEDIA && isPrintModelIcon()) ? ICON_PREVIEW : ICON_BABYSTEP;
    //gCreate testing to make preview thumb work
    //printingItems.items[KEY_ICON_5].icon = (infoFile.source < FS_ONBOARD_MEDIA && isPrintModelIcon()) ? ICON_BABYSTEP : model_DirectDisplay(getIconStartPoint(5), infoFile.path);
    printingItems.items[KEY_ICON_5].icon = ICON_PREVIEW;
```
[image]