bigtreetech/BIGTREETECH-TouchScreenFirmware #121: Multiple Icon Profiles

Sthopeless, 2019-11-07:
Hello everyone! 

Recently I see here and on facebook a lot of users releasing pack of icons for the TFT's.
And instead of having to 'reflash' everything I want to change icons pack, would it be possible to integrate them all in the firmware and choose which one to use directly on the TFT?

guruathwal, 2019-11-08:
That would need a lot of storage space.
The TFTs have very limited flash storage.

Hukuma1, 2019-11-08:
Is there a reason the icons aren't crushed PNGs instead of BMPs?

Did a quick test:
BackGroundColor.bmp = 28,672 bytes
BackGroundColor.png = 8,192 bytes

So in theory we could probably have 3X icons/different packs using the same space we do now with BMPs for 1 set.

Sthopeless, 2019-11-08:
cant the images be saved and load from the sdcard and only have loaded the package se choose?!

guruathwal, 2019-11-08:
> Is there a reason the icons aren't crushed PNGs instead of BMPs?
> 
> Did a quick test:
> BackGroundColor.bmp = 28,672 bytes
> BackGroundColor.png = 8,192 bytes
> 
> So in theory we could probably have 3X icons/different packs using the same space we do now with BMPs for 1 set.

Yes there are reasons.
First the Flash storage used in these devices is not like the normal SD card or USB drive. each file is stores as binary data in the form of bytes.
These flash storage do not have any file system to handle files.
The image data is read by specifying the starting byte address and the total byte counts with just few lines of coding. This is where BMP is used.
BMPs always have the same file size for specific dimensions. changing the color or design does not affect the size. In short BMP is uncompressed image.
Other formats like JPG, PNG, GIF etc. are compressed image format. File size will change if just one pixel is changed, making it difficult to track the data location for each image and will result in just garbage.

guruathwal, 2019-11-08:
> cant the images be saved and load from the sdcard and only have loaded the package se choose?!

The software will crash when you will remove the sd card for copying gcode files as there will be no storage to load data from.

Sthopeless, 2019-11-08:
> > cant the images be saved and load from the sdcard and only have loaded the package se choose?!
> 
> The software will crash when you will remove the sd card for copying gcode files as there will be no storage to load data from.

I see, but wouldn't be possible for example have all the icon packs in different named folders on the sdcard and on the TFT when we choose which pack to use the 'software' loads into it the necessary image?!

thisiskeithb, 2019-11-08:
> I see, but wouldn't be possible for example have all the icon packs in different named folders on the sdcard and on the TFT when we choose which pack to use the 'software' loads into it the necessary image?!

This is how I envisioned initial support for this feature could work (a select a pack from SD card and reboot to load it).

guruathwal, 2019-11-08:
> I see, but wouldn't be possible for example have all the icon packs in different named folders on the sdcard and on the TFT when we choose which pack to use the 'software' loads into it the necessary image?!

how hard can it be? may be possible like this:
1. copy icon packs in seperate folders (TFT35/Pack1 , TFT35/Pack2, TFT35/Pack3)
2. list all packs in a menu
3. select a pack
4. renames old selection to Pack4 (like "TFT35/BMP" to "TFT35/PACK4")
5. renames new selection to default icon folder(like "TFT35/PACK1" to "TFT35/BMP")
6. Reset to apply new pack.

thisiskeithb, 2019-11-08:
Now that I know PIO Core 4.0.3 can generate working firmware (4.1 won't), I will be able to work on this some more over the weekend.

I think improving the file selection interface would tie in nicely to this request.

Sthopeless, 2019-11-08:
@thisiskeithb on my TFT35 v2.0 I dont see a option to babystep if I am printing through Octoprint maybe we could have babystep somewhere all the time for when we are printing throught Octoprint.

maybe we could have have a few configurable pages that we could define which gcode each button sends, something like this:
[image]

Sthopeless, 2019-11-08:
Also awhile ago I saw this picture 
[image] couldn't something like this be done and enabled TFT35 v2.0 to also have the 12864 emulator? There should be enough pins available to connect to our boards

guruathwal, 2019-11-09:
> @thisiskeithb on my TFT35 v2.0 I dont see a option to babystep if I am printing through Octoprint maybe we could have babystep somewhere all the time for when we are printing throught Octoprint. [...]

I think @Msq001 is already working on it. He has added a custom option in the 'MachineSettings.c' in PR #91 , but it is not complete yet.
@Msq001 please confirm?

Msq001, 2019-11-09:
> I think @Msq001 is already working on it. He has added a custom option in the 'MachineSettings.c' in PR #91 , but it is not complete yet.
> @Msq001 please confirm?

yeah. The purpose is that user can customize the gcode to be sent and the name of the button in the configuration. h file. I'm doing this now,  And I will write some documents to explain the screen firmwre in detail including how to add languages, customize fonts, customize icons, and explain the effect of each option in configuration. h

Msq001, 2019-11-09:
> @thisiskeithb on my TFT35 v2.0 I dont see a option to babystep if I am printing through Octoprint maybe we could have babystep somewhere all the time for when we are printing throught Octoprint. [...]

I will add "BabyStep" in "MachineSetting", Thank you for your advice

Sthopeless, 2019-11-09:
That's all great news, thank you a lot guys!

Please think dearly about enabling 12864 emulation on other TFT's that have enough pins available. BTT Legend really deserves it.

Also is there any way to make the touch screen aware that the printer is printing or being used thought Octoprint?!

hapklaar, 2019-11-09:
> That's all great news, thank you a lot guys!
> 
> Please think dearly about enabling 12864 emulation on other TFT's that have enough pins available. BTT Legend really deserves it.
> 
> Also is there any way to make the touch screen aware that the printer is printing or being used thought Octoprint?!

That would be great and maybe a starting point for giving the TFT mode a proper 'live' status screen? Not sure if this is even possible, would love some insight from you guys as development for this TFT seems at an all-time high at the moment.

thisiskeithb, 2019-11-09:
[This Touch UI thread](https://github.com/MarlinFirmware/Marlin/issues/10008) would be a good read related to Marlin/LCDs/TFTs.

If there’s a way to help drive a new (better) standard, then I think it’d be worth pursuing.

Sthopeless, 2019-11-09:
If your printer only has touchscreen mode and you use Octoprint it can be very frustrating at times :)

Stop button on the TFT doesn't stop the print (only reset) and once turned off the heating of my hotend and bed and kept printing until the extruder started grinding on the dry filament but most of the time just stops homes and starts the print again from 0 on top of the old print.

Babystepping is always fun.. if you start the print from the tft sdcard (i never tried from the board sdcard) babystep function is there but is never happening on real time and if it is a big print sometimes I have to wait long time to see it changing, causing me sometimes to just have to restart the print. 
I cannot save the value so every time i reboot the printer is gone and most of times I forget to set it before starting the print..

Through octoprint the function is not there. there is now a gcode console but i think it still needs some improvements to be used

guruathwal, 2019-11-09:
> If your printer only has touchscreen mode and you use Octoprint it can be very frustrating at times :)
>
> Stop button on the TFT doesn't stop the print (only reset) and once turned off the heating of [...]

The stop button on TFT sends a emergecy stop Gcode "M112" to the printer. you need to enable `EMERGENCY_PARSER` for an instantaneous M112 command or else it will wait for the command  queue to clear.

Babystepping on TFT works by sending the gcode to the printer.  it will also wait for the command  queue to clear.

Sthopeless, 2019-11-09:
@guruathwal thanks! actually i did not know about having to enable ```EMERGENCY_PARSER``` for it to work properly!

About the babystepping I had that figured already, which makes me wonder if a display sending gcode that's queued in line is the 'way to go' for a viable display.

guruathwal, 2019-11-09:
> @guruathwal thanks! actually i did not know about having to enable `EMERGENCY_PARSER` for it to work properly!
> 
> About the babystepping I had that figured already, which makes me wonder if a display sending gcode that's queued in line is the 'way to go' for a viable display.

for the babystepping there should be hard wired buttons connected directly to the main board. i don't know why no one has tried to implement that.

Sthopeless, 2019-11-09:
sorry i dont think adding wired buttons is a solution, more as a 'hack around to get it working'

Wouldn't it be a better solution to 'redesign' the firmware's make it integrated with Marlin's (has a real LCD option we enable on Marlin) and make it only 1 firmware where we define if there is external rotary encoder or buttons in screen etc, the emulator doesn't need to be a emulator and all can be just one thing across all TFT's, Isn't something like this possible and more functional?

guruathwal, 2019-11-09:
Redesigning firmware for the TFT from almost scratch is a long way to go.

Sthopeless, 2019-11-10:
@guruathwal I understand that. But maybe inspiration can be taken from TFT's like from Prusa(?)

Anyhow I know that is a big shoot...

I gave some thoughts about the 'icons profiles' and maybe it could be done like this,
We have a folder on TFT SDCard for the icons profiles with a main picture for each profile and file name of the "Icon Pack"(maybe a preview of the icon pack) something like this:
[image]

Then everytime the TFT reboots would look into that folder see how many pictures/'icon packs' there are list them with the main picture name (without .bmp or whatever extension is used) and when clicked we'd see the main picture and a 2 options "cancel" which would go back to the icons pack select menu or appy that would reboot the tft and copy the necessary icons fonts etc into the memory 

'Main Picture' and 'Icons Pack folder' can maybe have the same name so the TFT knows which folder to copy.

This would also give us a option to also preview a little the packs before applying and would allow to add/remove more in the future without major changes in the firmware

Chaumezz, 2020-02-14:
Hola!   Me parece muy interesante este hilo, pero la comunicacion de la TFT siempre van a tener que pasar por el serial y no va a ser directamente sobre el manejo de la placa.
Me gustaria saber donde encontrar packs de iconos para la TFT35 para renovar su aspecto.
Gracias!

closed 2020-08-14 by bigtreetech