bigtreetech/BIGTREETECH-TouchScreenFirmware #122: No way to exit the Info screen

appideasDOTcom, 2019-11-07:
In touchscreen mode, if you tap Settings > Info, there's no way to return to the previous screen, so the only option is to hit the reset button.

Hardware: TFT24
Firmware: 1.1.24

This is obvious and 100% repeatable. Do I need to supply more info?

guruathwal, 2019-11-08:
You can touch anywhere on the screen to exit the info screen or disconnect screen.

appideasDOTcom, 2019-11-08:
Oh jeez. OK. I changed the subject, and this should be considered a featured request.

It's pretty strange to have one dangling "geek screen" (zero styling and completely different and non-intuitive controls) when everything else is supposed to be stylized for normal people to enjoy.

Presenting the information in a slightly nicer way and offering a BACK button should suffice.

guruathwal, 2019-11-08:
There is need to add this text " Touch any where to exit" at the end of the screen

appideasDOTcom, 2019-11-08:
That would suffice to avoid confusion, but it's still not a great user experience to go from icon-based navigation to a screen that's all black with a fixed-width font and no buttons.

thisiskeithb, 2019-11-08:
About screen behavior is a weird one. It's not uncommon for many applications to pop up a non-standard GUI/box with no buttons or obvious way to close them. Eventually, you'll click & it goes away.

appideasDOTcom, 2019-11-08:
I disagree that "It's not uncommon for applications to pop up a non-standard GUI/box with no buttons..." Even the nerdiest of nerd tools have a somewhat stylized About screen. I can't think of a single example of an unstylized About screen in any product I own, nor can I think of any example of such a screen that does not have a visible close control in the absence of a physical (or always-on virtual) button (think Android).

But it's not my decision, and adding text to the screen removes the confusion, at least. It just isn't a good user experience, and is totally inconsistent with every other screen in the interface.

I'll close this ticket since the this is being addressed on ticket #132 

Thank you!

closed 2019-11-08 by appideasDOTcom

thisiskeithb, 2019-11-08:
I was thinking desktop apps like Adobe Creative cloud, etc., but we have our own sample sizes of how common something is in the wild 🙂

appideasDOTcom, 2019-11-08:
Adobe Creative Cloud's About screen is EXTREMELY highly stylized. If there were such a thing as the opposite of a style, the Info screen on the TFT and Adobe's About screens are opposites. If you follow Adobe as an example, nobody will complain. That's not what's happening here. :-)

This is now a point of minutiae and not functionality, so I'll drop it.

Thank again!

thisiskeithb, 2019-11-08:
> Adobe Creative Cloud's About screen is EXTREMELY highly stylized

I didn’t say it had to be non-stylized. I said:

> It's not uncommon for applications to pop up a non-standard GUI/box with no buttons or obvious way to close them.

Which describes the Adobe about screen and closing behavior accurately.

guruathwal, 2019-11-10:
for comparison
about screen, feature rich VSCodium have a very simple about screen and the simpler Arduino IDE have a beautiful designed about screen.
in contrast to this in Arduino IDE there is no close button or any type of instruction to close it. you have to click anywhere on it to close and The VSCodium about screen have a close button.

appideasDOTcom, 2019-11-10:
I'm dropping it. :-) Thank you!

I appreciate all the work everyone is doing to make these great products better. I don't want my appreciation to be unknown amidst the little complaints I may have from time to time.