bigtreetech/BIGTREETECH-TouchScreenFirmware #123: TFT24 icons do not change

appideasDOTcom, 2019-11-07:
Hardware: TFT24 1.1
Connected to: SKR V1.3 (probably irrelevant)
Firmware: 1.1.24

Attempting to change icons has no effect. What I did:
1. Copy the contents of "Copy to SD Card root directory to update/TFT24" to an SD card
2. Load the SD card into the TFT24
3. Hit the reset button

The TFT goes through the process of updating icons and shows the icons that it thinks it is replacing, but when it comes to my custom icons, it displays the old icon instead of the new.

As a test, I reduced the files in the TFT24 directory to only Logo.jpg and marlin.jpg. The only image it displayed during the icon update process was the one for marlin, and what it displayed on the screen was the original file, not the one I copied to the card.

I formatted the SD card to be sure that it wasn't pulling the old image from somewhere I wasn't seeing.

Either the icon updater is broken or it is not accepting the format of my BMP files. In either case, I have no clue where it is getting the old icon for the update since it isn't on the SD card.

One other fun detail is that after attempting to update the icons and rebooting the controller, the remaining TFT24.CUR folder on the SD card has my custom icons.

guruathwal, 2019-11-08:
Are you sure your icon are in this directory:
/TFT24/bmp/
in yor sd card
All your icon need to be in bmp folder.
You can not update icons from multiple folders

appideasDOTcom, 2019-11-08:
Yes. I am absolutely, 100% certain. The attached screenshot shows the contents of the SD card before hitting the reset button. After I hit the reset button with the SD card inserted, the screen goes through an update process, shows icons that it is updating, but displaying the original version of the files I am trying to replace. When I insert the SD card into a computer afterward, the TFT24 directory has been renamed TFT24.CUR
[image]

I have also:
- forcibly reduced the color palette of each file to 8 bits
- tried 70px x 70px icons to match those from the material theme folder (after trying the standard 95x95 and failing)
- reduced images to 8 bit
- reduced density to 72 dpi (although some provided by bigtree were different)
- overwrite default images while loading the stock firmware file from the github repository instead of the custom one I've compiled after changes to Configuration.h
- only overwrite the images and not the firmware or font
- only overwrite the firmware and the images
- only overwrite the images and font

Unless my screen is faulty, I can't imagine anything on my end that I could have possibly missed that would cause it to read through the TFT24/bmp directory and still use the original files, but I have proof that it is, in fact, reading from the TFT24/bmp directory.

I have seen what happens when you only include files you want updated in the TFT24/bmp directory instead of including every file - it only tries to update the files you include. I have witnessed that and watched it say "marlin.bmp" while showing the original image (not mine) and no other files. So I know that it is recognizing the directory and specific file names that are present. It's simply failing to update them.

guruathwal, 2019-11-08:
- The Images should be 24bit.
- 70x70 or 95x95 depends on your screen size. for TFT35 use 95x95 and for TFT24/TFT28 use 70x70.
- DPI does not do anything on Displays, it is used for printing.

you need to re-format you SDcard to FAT32, then copy the files.

appideasDOTcom, 2019-11-08:
It seems that FAT32 was the thing. The SD card was previously formatted with ExFat, which I've never had an issue with. What's confusing about that is the fact that it clearly loaded new firmware and was able to tell which files I was trying to replace. I have been using the same card formatted with ExFat to load a now heavily modified firmware, and everything was working except the icons.

I had started with 24 bit images, but reduced them to 8 bit in an attempt to simplify everything and eliminate possible causes.

In tracking this down, I found a definite bug. I'll close this ticket and open a new one.

Thank you!

closed 2019-11-08 by appideasDOTcom

appideasDOTcom, 2020-02-10:
For anyone who comes across this issue and the format of the SD card was not the issue, I have found another scenario that gives the same symptoms: When I exported the BMP files from Pixelmator or Preview, I had the same symptoms. I opened the BMP files that those apps created in GIMP and exported them as BMP without changing any options, and the files updated correctly on the TFT.

Apparently it's picky about the format of the SD card as well as something internal to the BMP files.

deancowan, 2020-09-09:
I just tried updating my Logo.bmp as well as a the most recent firmware updated for my TFT35 and was brought to this page when my logo didn't update. Turns out, paint.net saved my edited BMP only as 8 bit (256 colours) based on the simple black/white image. I re-edited in MS Paint and saved as 24 bit. Worked a treat! Hope this helps others.

Side note: The Aug 27 2020 firmware seems to be asking for a missing EXTRUDE_100MM.BMP. Strange, the bmp file doesn't exist but I can't see where it's needed (certainly doesn't affect the 100mm extrude icon)

[image]