bigtreetech/BIGTREETECH-TouchScreenFirmware #124: Touchscreen menus/options need to be consistent

appideasDOTcom, 2019-11-07:
As it is, touchscreen mode is only half usable. For example, If I turn on the "Unified Menu" I don't have any preheat options. If I turn off the Unified Menu, I lose the buttons that give me quick access to my defined manual leveling points.

I have come across at least three useful features that are available in one menu system and not the other, which has lead me to abandon touchscreen mode and revert to the "old" mode.

It isn't acceptable to have the available capabilities determined by choice of menu structure. If a feature exists, users need to be able to get to it.

UnknownHero99, 2019-11-07:
it was changed that manual leveling points are available on normal menu only if auto bed leveling is disabled in configuration file, so it's a choice between auto bed leveling and manual bed leveling and about preheat menu it should be added when merging, i just added pull request #125 and it should be there after it's applied

appideasDOTcom, 2019-11-07:
Awesome. My additional suggestion would be to add a comment to Configuration.h letting people know that they need to disable ABL in order to enable manual bed leveling options (is "MBL" a valid acronym?). Thank you!

closed 2019-11-07 by appideasDOTcom

AnHardt, 2019-11-07:
MBL could be confused with Marlins "**M**esh **B**ed **L**eveling".