bigtreetech/BIGTREETECH-TouchScreenFirmware #126: TFT35 V2 latest version not updating

UnknownHero99, 2019-11-07:
when trying to update to the latest version it just stucks at 100%, earlier version works just fine

guruathwal, 2019-11-08:
What changes have you made to your code?

thisiskeithb, 2019-11-08:
I'm experiencing a similar issue with a TFT35 V3 with self-compiled firmware after the changes just implemented. Precompiled [from a couple days ago works fine](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/2829e446ec5cccd5fe0679ec6e17f7a1885cb4de/Copy%20to%20SD%20Card%20root%20directory%20to%20update/BIGTREE_TFT35_V3.0.24.2.bin).

guruathwal, 2019-11-08:
Try pressing reset button after of it reaches 100.
If it boots check the build number and date to confirm if the firmware updated or not.

thisiskeithb, 2019-11-08:
The binary is loaded on reset, but the LCD stays blank after the initial load like it’s dead. And resetting (without the SD card) does nothing.

Loading the older binary I linked gets it working again.

I’ll see if downgrading PIO fixes it since recent commits shouldn’t have caused this issue.

UnknownHero99, 2019-11-08:
@guruathwal  no changes at all the problem is exactly the same as @thisiskeithb is saying, removing sd card and pressing reset button only shows a black screen, prebuilt works just fine

@thisiskeithb this surely could be due to PIO update didn't even notice it updated

guruathwal, 2019-11-08:
> @guruathwal no changes at all the problem is exactly the same as @thisiskeithb is saying, removing sd card and pressing reset button only shows a black screen, prebuilt works just fine
> 
> @thisiskeithb this surely could be due to PIO update didn't even notice it updated

the current PIO version on my PC is : **Home 2.3.3·Core 4.0.3** running on VSCodium **1.36.1**, with which the firmware update is working fine.
what is your PIO version?

thisiskeithb, 2019-11-08:
4.1 was just released, so that’s what I’m running. I can’t downgrade right now, but will be able to test this weekend.

UnknownHero99, 2019-11-08:
I can confirm PIO is causing it. I just downgraded from 4.1 -> 4.0.3 and it's working fine now.
looks like there is a bug in 4.1 PIO Core so until they fix it don't upgrade to 4.1, if u already did 
`pip install "platformio==4.0.3" --force-reinstall --user` does the trick

thisiskeithb, 2019-11-08:
From the [PIO CEO](https://github.com/MarlinFirmware/Marlin/issues/15767#issuecomment-551741299) in a related issue:

> Could someone provide me with AnyDesk remote session? I would be thankful to debug this issue and find which commit has a problem. There are over 200 commits to PIO Core 4.1 since 4.0.3 and 2.5 months of work. It is very difficult to find a problem.
> 
> Please contact me@ikravets.com with a session ID. Thanks!

ivankravets, 2019-11-08:
Please merge this https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/131

Also, please don't use internal/temporary variables from PIO Core build system. There are public scopes and we recommend using them.

See http://docs.platformio.org/en/latest/projectconf/section_env_build.html#build-flags

thisiskeithb, 2019-11-08:
@bigtreetech: Please merge this 🙂

thisiskeithb, 2019-11-08:
@ivankravets: Still no dice with your latest changes. Same behavior as before with blank screen/no boot. Restoring to the previous binary (compiled pre 4.1) brings it back to life.

[Link to my TFT35 firmware with your changes](https://github.com/thisiskeithb/BIGTREETECH-TouchScreenFirmware/tree/7a27c03f876b0ec7c5241f4c004f045ee3a1b0bb).

ivankravets, 2019-11-08:
I've just re-tested and it builds the same firmware as PIO Core 4.0.3. 
Could you run `pio run -t clean` and start building project again? Or, delete `.pio` folder form a project.

thisiskeithb, 2019-11-08:
> `pio run -t clean`

I already run that before every build out of habit, but ran it again & also manually deleted the .pio folder.

While compiles successfully, the hardware just don't work compiling with 4.1. Going back to the binary generated pre 4.1, it updates & boots back up as expected.

ivankravets, 2019-11-08:
Could you use the same code but change `platform      = ststm32` to `platform      = ststm32@5.6.0` ?

thisiskeithb, 2019-11-08:
This is probably a clue:

> Warning! Cannot find a linker script for the required board! Firmware will be linked with a default linker script!

ivankravets, 2019-11-08:
> This is probably a clue:

It's OK, false warning. We will fix it soon.

thisiskeithb, 2019-11-08:
> Could you use the same code but change platform = ststm32 to platform = ststm32@5.6.0 ?

No change. Screen still appears dead.

guruathwal, 2019-11-08:
did you tried clean install of PIO?

thisiskeithb, 2019-11-08:
> did you tried clean install of PIO?

No, just an upgrade via PIO's built-in check. I'm running PIO Core 4.1 on one machine and 4.0.3 on another so I don't have to deal with upgrading/downgrading.

thisiskeithb, 2019-11-08:
I can build for LPC1768 without issue in both environments.

hapklaar, 2019-11-09:
> > Could you use the same code but change platform = ststm32 to platform = ststm32@5.6.0 ?
> 
> No change. Screen still appears dead.

I can confirm. Building with Pio 4.1 results in a firmware that gives an empty/dead screen after boot. After downgrading to 4.0.3 it works again.

ivankravets, 2019-11-09:
Could someone re-test this PR https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/136?

I don't have this hardware.

thisiskeithb, 2019-11-09:
> Could someone re-test this PR #136?
> 
> I don't have this hardware.

It works!

ivankravets, 2019-11-09:
Great! Let me create a similar PR to Marlin repository.

Thanks for testing!

ivankravets, 2019-11-09:
@bigtreetech could you merge https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/136 ?

hapklaar, 2019-11-09:
> > Could someone re-test this PR #136?
> > I don't have this hardware.
> 
> It works!

Can also confirm :) Thanks!

thisiskeithb, 2019-11-09:
Thank you for the assistance @ivankravets (and PIO) 🙂

UnknownHero99, 2019-11-09:
Since the issue will be then @bigtreetech merges #136 I'm closing this issue

closed 2019-11-09 by UnknownHero99