bigtreetech/BIGTREETECH-TouchScreenFirmware #128: BigTree logo improperly named for icon replacement

appideasDOTcom, 2019-11-08:
The updateIcon() function in src/User/API/boot.c is iterating the iconBmpName array, which contains strings matching file names to look for on an SD card for replacing the device's icons.

According to the iconBmpName array, the file "BigTreeTech.bmp" should be present in the bmp directory, but no such file exists in the repository. 

What people who haven't examined the source code would expect to work: The closest file in the bmp directory is "bigtree.bmp" but replacing that file has no effect - the BigTree logo still appears on the mode selection screen (the screen you get when you hold the physical button for three seconds).

Confirmed workaround: Adding the file "BigTreeTech.bmp" to the SD card replaces the on-screen icon as expected.

What should be done: Rename bigtree.png in the repository to BigTreeTech.bmp

guruathwal, 2019-11-08:
🤔

[image]
 
1 word:
**INCORRECT**

2 words:
**CHECK AGAIN**

3 words:
**UPDATE YOUR REPOSITORY**

4 words:
**DO NOT CRITICIZE PEOPLE**

5 words:
**YOU SHOULD CLOSE THE ISSUE**

PS:
Please do not criticize people. it takes a lot of time and hard work to code. even if there was a mistake you could have politely point it out. No one is getting paid here to share their contributions to the source. If everyone starts criticizing each other for mistakes then on one will be here. 
I hope you will understand that.

Tim-The-Woodsman, 2019-11-08:
I fail to find the harsh criticism. I agree to the rest. What are you getting angry about?

thisiskeithb, 2019-11-08:
As @guruathwal pointed out, BigTreeTech.bmp is present in both of the current icon packs as well as each TFT variant folder.

This was fixed in https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/100 four days ago, so update to the latest firmware and everything will work.

appideasDOTcom, 2019-11-08:
Sorry. I had failed to update. :-/ Bad developer! Check the repo! Closing the ticket.

I'm not sure where you saw criticism in my report. I believed that I was simply stating facts that I had observed and was relaying them as facts. I apologize if it came off as criticism. I definitely appreciate the hard work of the folks at bigtree, and appreciate even more the work they're doing to push the quality of low price 3D printer electronics.

closed 2019-11-08 by appideasDOTcom