bigtreetech/BIGTREETECH-TouchScreenFirmware #133: Stop printing on TFT35 v3.0 V1.0.24.1.bin, V1.0.24.2.bin.

Aleks-MO, 2019-11-08:
Printing from SD or Flash card stops at any time. If you print from a card to SKR, after printing is stopped, there is no TFT connection with SKR.  ?????

UnknownHero99, 2019-11-08:
do you use any other device that is connected to SKR's serial connection? , like PC, ESP etc..

Aleks-MO, 2019-11-08:
No PC, no ESP.  https://www.youtube.com/watch?v=k0uLvApJYCE

Aleks-MO, 2019-11-08:
@UnknownHero99  Flash stop.  https://www.youtube.com/watch?v=vN83fgP5zWg

guruathwal, 2019-11-09:
Try printing the same gcode file through PC without the TFT and see if it prints fine.

Aleks-MO, 2019-11-09:
@guruathwal  prints in 12864, the file has nothing to do with it, the COM connection breaks!

guruathwal, 2019-11-09:
#129 addressed some bugs in printing from onboard sdcard.
Please check if it solves your issue.

Aleks-MO, 2019-11-09:
@guruathwal No, it's still a loss of communication. I’m doing a screen reset, I see an SD card, I click print, I stop and I don’t take everything.

bigtreetech, 2019-11-09:
> @guruathwal No, it's still a loss of communication. I’m doing a screen reset, I see an SD card, I click print, I stop and I don’t take everything. I increase the speed to 250,000 the problem remains.

hello,  if you print from onboard sd card,  There will be an M109 S200 at the start of gcode to set the target temperature. After Marlin executes M109, Marlin will refuse to execute all the gcodes received by the serial port. Marlin will not response the serial port communication until the actual temperature reaches the set temperature by M109.   so you must wait for the temperature heated to 200 degrees. then the communicate will resume.

Aleks-MO, 2019-11-09:
@bigtreetech  
we are about board with smoothieware.

Hukuma1, 2019-11-09:
@bigtreetech Does this mean we will not see the temperature rise on our touch screen mode? Currently trying to print off onboard mSD as a test and it just sits there. Did not seem to be heating the bed or anything either. Using latest self compiled version from the commits made yesterday.

Aleks-MO, 2019-11-14:
@bigtreetech  Who can see the transfer code on RX TX to the board from the screen. The stop is there, the loss of the transmission packet.

Baltazart, 2019-11-14:
> @bigtreetech Who can see the transfer code on RX TX to the board from the screen. The stop is there, the loss of the transmission packet.

sorry for my English
I read in the title "Stop printing on TFT35 v3.0 V1.0.24.1.bin, V1.0.24.2.bin" Are you sure that it is compatible with this screen? normally "BIGTREE_TFT35 V3.0.24.2.bin (v3)".
do you confirm that you have installed V1.0.24.1.bin? .

Aleks-MO, 2019-11-14:
@Baltazart  I'm not a native English speaker either. Yes, there is an error, read correctly, 3,0,24,2.bin.  The stop of printing comes from the loss of the exchange packet, this is my opinion. But I not so well know that there programmatically implemented.

closed 2020-08-14 by bigtreetech