bigtreetech/BIGTREETECH-TouchScreenFirmware #137: TFT35 v3 Not printing from USB anymore

Hukuma1, 2019-11-09:
Compiled latest merges as of Nov 8th 2019 (PIO 4.0.3 not 4.1.0) (confirmed with new preheat icons working)

When I insert USB drive it detects it (U disk inserted!) but as soon as I click to access it, it gets stuck in some loading loop. Screen flashes really quickly and clearly in error. Cannot print. Prior versions have been working fine, so it has to be some change made to USB. Any ideas?

thisiskeithb, 2019-11-09:
Just tried 2a3f376 rebased on top of the pending PIO Core 4.1 fixes and I am seeing the same thing on a TFT35 V3.

thisiskeithb, 2019-11-09:
This is related to changes in https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/127 and is a bug.

To get around it for now without code changes, you have to include a bitmap with the same name as your `.gcode` file, but with a `_95.bmp` or `_70.bmp` file name end/extension in the same folder.

Example: If you have `PLA-benchy.gcode` on your USB drive, you also need to have `PLA-benchy_95.bmp` for the TFT35 or `PLA-benchy_70.bmp` for TFT28 & TFT24.

The preview doesn't work on my setup using one of the existing bitmaps as a test, but at least it doesn't crash 🙂

thisiskeithb, 2019-11-09:
After some more testing, image previews will show up if printing from SD card, but not USB:

[image]

Hukuma1, 2019-11-09:
How did you end up getting it to display? I tried creating a 95x95 bmp and it still failed to load.

bdaylitho.gcode
bdaylitho_95.bmp

Just shows regular gcode icon. Trying it off the SD card.

thisiskeithb, 2019-11-09:
Photoshop’s default BMP save options work for me, but [here’s a customization guide](https://github.com/bigtreetech/BIGTREETECH-TFT35-V1.2/blob/master/TFT35%20screen%20customize.pdf) with format details.

Hukuma1, 2019-11-09:
Thanks! I had tried OS/2 for some reason for BMP. Oops. :)

Works nicely. I know MKS had a Cura plugin to automate the image creation (https://github.com/makerbase-mks/Software/tree/master/MKS%20Plugin), but it seems to embed the preview into the gcode instead of an image along side it.

Wonder if BTT will have a similar approach or a means of automating the preview creation?

thisiskeithb, 2019-11-09:
Embedding images in gcode? Yikes.

The [BIQU Tool](https://github.com/bigtreetech/BiQU_Tool) will help with preview generation, but I’m not sure what BTT’s long-term plans are for this product lineup.

Hukuma1, 2019-11-10:
Oh geeze. I hope they move over to a plugin approach. It took my 3900X almost 2 minutes to even render my litho stl.

No one is going to take the time to go through this workflow. Even easier to just screenshot out from Cura and crop in Photoshop. D:

Anyway, thanks for finding the bug for USB drives.

Hukuma1, 2019-11-11:
Latest merges fix this for me. Closing. Thank you!

closed 2019-11-11 by Hukuma1