bigtreetech/BIGTREETECH-TouchScreenFirmware #138: Custom extrusion length and motor movement

Sthopeless, 2019-11-10:
Is it possible to add 100 to the options (atm we have 1, 5 and 10) to move motors and load/unload filament and a 5th one we would define the amount in configuration.h maybe leave default around 400's since is the amount for bowden systems (?)

thisiskeithb, 2019-11-10:
Making them configurable wouldn’t be difficult.

 Marlin’s default `EXTRUDE_MAXLENGTH` is 200, so that means it’ll be the default for many printer configs which can be a potential issue.

Sthopeless, 2019-11-10:
@thisiskeithb, yes you are right, I forgot the default on Marlin's is 200 and most people don't change it. 
Perhaps than could be added 100 and 200 option and the 200 could be configurable on configuration.h with a note explaining it will limited by Marlin's ```EXTRUDE_MAXLENGTH``` on line 542

Hukuma1, 2019-11-10:
I am waiting for the proper documentation of the custom functions just for this. This way we can map a button to unload 430 filament and then load it back up once I switch it out. Right now I have to keep hitting the extrude button for manual operation. Hopefully we easily macro commands to one button. (so we can send 2 commands of extrude 200 to still comply with Marlin's default max?)

Sthopeless, 2019-11-10:
@Hukuma1 you can recompile Marlin to allow 430 of filament look for ```EXTRUDE_MAXLENGTH``` in configuration.h and change that value to whatever you want (430 as you mentioned).

This way I am requesting we could have 100 and a configurable value added to the menus instead of using one of the future custom buttons and it would be nested where it should

thisiskeithb, 2019-11-10:
There's some work by @Msq001 to load values from .gcode, so this feature could be flexible.

closed 2020-08-14 by bigtreetech